raspberrypi/Raspberry-Pi-OS-64bit #80: Not able to install wireguard

MidGe48, 2020-07-31:
Not able to install wireguard due to missing linux-headers on Pi4 4Gb.

pelwell, 2020-07-31:
Duplicate of #4 .

scito, 2020-10-24:
I've manged to compile it using the info of #4 

```bash
# This library is needed when running 'make menuconfig'
sudo apt install libncurses5-dev
sudo apt-get install rpi-source
# set arm_64bit=0 in /boot/config.txt
# see https://github.com/RPi-Distro/rpi-source
sudo rpi-source
sudo dpkg-reconfigure wireguard-dkms
dkms status wireguard
# set arm_64bit=1 in /boot/config.txt
```
My output is
```
wireguard, 1.0.20200908, 5.4.72+, armv7l: installed
wireguard, 1.0.20200908, 5.4.72-v7+, armv7l: installed
wireguard, 1.0.20200908, 5.4.72-v7l+, armv7l: installed
wireguard, 1.0.20200908, 5.4.72-v8+, armv7l: installed
```

But I cannot start wireguard in 64-bit mode (having set arm_64bit=1 in /boot/config.txt)
```
# wg-quick up wg0
[#] ip link add wg0 type wireguard
Error: Unknown device type.
Unable to access interface: Protocol not supported
[#] ip link delete dev wg0
Cannot find device "wg0"
```

The wireguard kernel module is not started and it cannot be started manually:
```
# modprobe wireguard
modprobe: ERROR: could not insert 'wireguard': Exec format error
```

In summary, it seems wireguard is compiling for armv8+, but it is not working yet.

lurch, 2020-10-24:
> wireguard, 1.0.20200908, 5.4.72-v8+, armv7l: installed
modprobe: ERROR: could not insert 'wireguard': Exec format error

Looks like you've installed an armv7l (32-bit) kernel module into your armv8 (64-bit) kernel modules directory?

scito, 2020-10-24:
> Looks like you've installed an armv7l (32-bit) kernel module into your armv8 (64-bit) kernel modules directory?

Yeah, probably.
I was just executing `sudo dpkg-reconfigure wireguard-dkms` and then, see output below.

```
$ sudo dpkg-reconfigure wireguard-dkms

------------------------------
Deleting module version: 1.0.20200908
completely from the DKMS tree.
------------------------------
Done.
Loading new wireguard-1.0.20200908 DKMS files...
It is likely that 5.4.72-v7l+ belongs to a chroot's host
Building for 5.4.72+, 5.4.72-v7+, 5.4.72-v7l+ and 5.4.72-v8+
Building initial module for 5.4.72+
Done.

wireguard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72+/kernel/net/

depmod...

DKMS: install completed.
Building initial module for 5.4.72-v7+
Done.

wireguard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72-v7+/kernel/net/

depmod...

DKMS: install completed.
Building initial module for 5.4.72-v7l+
Done.

wireguard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72-v7l+/kernel/net/

depmod...

DKMS: install completed.
Building initial module for 5.4.72-v8+
Done.

wireguard.ko:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/5.4.72-v8+/kernel/net/

depmod....

DKMS: install completed.
```

Aftwards the output is
```
$ dkms status wireguard
wireguard, 1.0.20200908, 5.4.72+, armv7l: installed
wireguard, 1.0.20200908, 5.4.72-v7+, armv7l: installed
wireguard, 1.0.20200908, 5.4.72-v7l+, armv7l: installed
wireguard, 1.0.20200908, 5.4.72-v8+, armv7l: installed
```

Is there a way to fix it?

lurch, 2020-10-24:
I'm afraid I know nothing about dkms, but at a rough guess I think you'd need to tell it to not build `5.4.72-v8+` when you're running a 32-bit kernel, and to _only_ build 5.4.72-v8+ when you're running a 64-bit kernel? (which also assumes that dkms uses the currently-running-arch when deciding what arch to build for).

Perhaps running a mixture of 32-bit and 64-bit is "asking for trouble", and it's best to stick to a pure 64-bit only install? :man_shrugging:

scito, 2020-10-25:
> it's best to stick to a pure 64-bit only install? man_shrugging

Agree, but then the build of wireguard fails and I am not able to install gcc aarch64/arm64.

```
$ sudo dpkg-reconfigure wireguard-dkms

-------- Uninstall Beginning --------
Module:  wireguard
Version: 1.0.20200908
Kernel:  5.4.72+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

wireguard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.72+/kernel/net/
rmdir: failed to remove 'kernel/net': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod....

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  wireguard
Version: 1.0.20200908
Kernel:  5.4.72-v7+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

wireguard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.72-v7+/kernel/net/
rmdir: failed to remove 'kernel/net': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  wireguard
Version: 1.0.20200908
Kernel:  5.4.72-v7l+ (armv7l)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

wireguard.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.4.72-v7l+/kernel/net/
rmdir: failed to remove 'kernel/net': Directory not empty
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  wireguard
Version: 1.0.20200908
[... 57 lines cut ...]
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1) 
```

```
$ sudo apt-get install crossbuild-essential-arm64
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 crossbuild-essential-arm64 : Depends: gcc-aarch64-linux-gnu (>= 4:8.3) but it is not installable or
                                       gcc:arm64 but it is not installable
                              Depends: g++-aarch64-linux-gnu (>= 4:8.3) but it is not installable or
                                       g++:arm64 but it is not installable
E: Unable to correct problems, you have held broken packages.
```

lurch, 2020-10-25:
Sorry, just to clarify - I meant starting from scratch with a fresh install of https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/ rather than having a weird mix of 32-bit and 64-bit stuff, as you seem to have in your current install.

It's beginning to feel like this is a problem more suited to https://www.raspberrypi.org/forums/ than the Raspberry-Pi-OS-64bit issue tracker?

fl4co, 2020-10-26:
I have the same problem as lurch on a pure 64 bit installation. Up until the latest raspberrypi-kernel-headers the module was building just fine.

lurch, 2020-10-26:
> I have the same problem as lurch...

**I** don't have a problem, I was only trying to suggest solutions or workarounds :wink: I've never actually tried to install or use wireguard myself.

> on a pure 64 bit installation. Up until the latest raspberrypi-kernel-headers the module was building just fine.

Weird, https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/4#issuecomment-702452905 suggests that dkms is working fine with the new kernel-headers package. If it's definitely the raspberrypi-kernel-headers package that is causing problems, could you post more details in #4 please?

XECDesign, 2020-10-26:
It's not going to work with a 32bit userland and 64bit kernel. There's no aarch64 compiler you can use to build the module.

usefitsoftware, 2021-12-26:
i had the same issue, solved with sudo apt-get update
check more info here:
https://forums.raspberrypi.com/viewtopic.php?t=268239