raspberrypi/Raspberry-Pi-OS-64bit #81: Not able to install zeek or zeek-lts

MidGe48, 2020-07-31:
Not able to install zeek or zeek-lts due to missing dependencies on Pi3B+

lurch, 2020-07-31:
According to https://packages.debian.org/search?keywords=zeek&searchon=names&suite=stable&section=all there's no "zeek" package in Debian?

rag2, 2020-07-31:
Michel,

the binary package at 
https://software.opensuse.org//download.html?project=security%3Azeek&package=zeek-lts, 
which is presumably what you tried t install, is an *armhf* package, not 
an *aarch64* (arm64) package. You may get better results using the 
*multiarch* approach (see https://wiki.debian.org/Multiarch/HOWTO).

You may be better off compiling from source, since if you need an IDS, 
it needs to be bullet-proof.

HTH,

Robin

> Debian's *multiarch*system^1 <#sdfootnote1sym> provides asolution^2 
> <#sdfootnote2sym> by allowing an *armhf* capability to be added, as an 
> additional supported architecture.
>
> sudo dpkg --add-architecture armhf
>
> To resolve an error around any package, usually a library package, 
> requires installing the *armhf*version of that package, denoted by an 
> *:armhf*suffix. If *libX11.so.6*, for example, is required:
>
> sudo apt install libx11-6:armhf
>
> 1 <#sdfootnote1anc>See https://wiki.debian.org/Multiarch/HOWTO.
>
> 2 <#sdfootnote2anc>Adapted from a description at 
> https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/62#issuecomment-647459572.
>