raspberrypi/Raspberry-Pi-OS-64bit #82: Anydesk installation

nickpar, 2020-08-03:
while Anydesk installs just fine on 32 bit

https://anydesk.com/en/downloads/raspberry-pi

we are unable to install on 64 bit

pelwell, 2020-08-03:
In what way does it not install or not work?

nickpar, 2020-08-03:
while the following package works fine on 32bit here in detail is what i get with this 64 bit version

https://anydesk.com/download?os=raspi

anydesk_5.5.5-1_armhf.deb

$ sudo dpkg -i anydesk_5.5.5-1_armhf.deb 
Selecting previously unselected package anydesk:armhf.
(Reading database ... 90789 files and directories currently installed.)
Preparing to unpack anydesk_5.5.5-1_armhf.deb ...
Failed to stop anydesk.service: Unit anydesk.service not loaded.
Unpacking anydesk:armhf (5.5.5) ...
dpkg: dependency problems prevent configuration of anydesk:armhf:
 anydesk:armhf depends on libglib2.0-0 (>= 2.16.0).
 anydesk:armhf depends on libgtk2.0-0 (>= 2.20.1).
 anydesk:armhf depends on libstdc++6 (>= 4.1.1).
 anydesk:armhf depends on libx11-6.
 anydesk:armhf depends on libxcb-shm0.
 anydesk:armhf depends on libxcb1.
 anydesk:armhf depends on libpango1.0-0.
 anydesk:armhf depends on libcairo2.
 anydesk:armhf depends on libxrandr2 (>= 1.3).
 anydesk:armhf depends on libx11-xcb1.
 anydesk:armhf depends on libxtst6.
 anydesk:armhf depends on libxfixes3.
 anydesk:armhf depends on libxdamage1.
 anydesk:armhf depends on libxkbfile1.
 anydesk:armhf depends on libegl1-mesa.
 anydesk:armhf depends on libgl1.

dpkg: error processing package anydesk:armhf (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 anydesk:armhf

i clean up :

$ sudo apt-get --fix-broken install
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following packages will be REMOVED:
  anydesk:armhf
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 10.9 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 90807 files and directories currently installed.)
Removing anydesk:armhf (5.5.5) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for desktop-file-utils (0.23-4) ...

and i wonder if there is dependency issue or just 64 bit

nickpar, 2020-08-03:
* Needed dependencies on Linux: libc6 (>= 2.7), libgcc1 (>= 1:4.1.1), libglib2.0-0 (>= 2.16.0), libgtk2.0-0 (>= 2.20.1), libstdc++6 (>= 4.1.1), libx11-6, libxcb-shm0, libxcb1, libpango1.0-0, libcairo2, libxrandr2 (>= 1.3), libx11-xcb1, libxtst6, libxfixes3, libxdamage1 and libgtkglext1. 

https://anydesk.com/en/downloads/raspberry-pi?os=raspi

nickpar, 2020-08-03:
checked the list above and i am unsure there is something missing

but hold on

 $ sudo apt-get install libglib2.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libglib2.0-cil-dev' for regex 'libglib2.0'
Note, selecting 'libglib2.0-tests' for regex 'libglib2.0'
Note, selecting 'libglib2.0-0-dbg' for regex 'libglib2.0'
Note, selecting 'libglib2.0-dev-bin' for regex 'libglib2.0'
Note, selecting 'libglib2.0-bin' for regex 'libglib2.0'
Note, selecting 'libglib2.0-cil' for regex 'libglib2.0'
Note, selecting 'libglib2.0-dev' for regex 'libglib2.0'
Note, selecting 'libglib2.0-doc' for regex 'libglib2.0'
Note, selecting 'libglib2.0-data' for regex 'libglib2.0'
Note, selecting 'libglib2.0-0' for regex 'libglib2.0'
libglib2.0-0 is already the newest version (2.58.3-2+deb10u2).
libglib2.0-0 set to manually installed.
libglib2.0-bin is already the newest version (2.58.3-2+deb10u2).
libglib2.0-bin set to manually installed.
libglib2.0-data is already the newest version (2.58.3-2+deb10u2).
libglib2.0-data set to manually installed.
The following additional packages will be installed:
  binfmt-support ca-certificates-mono cli-common libblkid-dev libffi-dev libmono-btls-interface4.0-cil libmono-corlib4.5-cil libmono-i18n-west4.0-cil
  libmono-i18n4.0-cil libmono-security4.0-cil libmono-system-configuration4.0-cil libmono-system-core4.0-cil libmono-system-numerics4.0-cil
  libmono-system-security4.0-cil libmono-system-xml4.0-cil libmono-system4.0-cil libmount-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5
  libselinux1-dev libsepol1-dev mono-4.0-gac mono-gac mono-runtime mono-runtime-common mono-runtime-sgen uuid-dev
Suggested packages:
  monodoc-gtk2.0-manual libxml2-utils devhelp libmono-i18n4.0-all libgamin0
The following NEW packages will be installed:
  binfmt-support ca-certificates-mono cli-common libblkid-dev libffi-dev libglib2.0-cil libglib2.0-cil-dev libglib2.0-dev libglib2.0-dev-bin
  libglib2.0-doc libglib2.0-tests libmono-btls-interface4.0-cil libmono-corlib4.5-cil libmono-i18n-west4.0-cil libmono-i18n4.0-cil
  libmono-security4.0-cil libmono-system-configuration4.0-cil libmono-system-core4.0-cil libmono-system-numerics4.0-cil libmono-system-security4.0-cil
  libmono-system-xml4.0-cil libmono-system4.0-cil libmount-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libselinux1-dev libsepol1-dev
  mono-4.0-gac mono-gac mono-runtime mono-runtime-common mono-runtime-sgen uuid-dev
0 upgraded, 35 newly installed, 0 to remove and 0 not upgraded.
Need to get 13.5 MB of archives.
After this operation, 70.4 MB of additional disk space will be used.
Do you want to continue? [Y/n] 

installing now

nickpar, 2020-08-03:
so all are here now

$ sudo apt-get install  libc6 libgcc1 libglib2.0-0 libgtk2.0-0  libstdc++6 libx11-6 libxcb-shm0 libxcb1 libpango1.0-0 libcairo2 libxrandr2 libx11-xcb1 libxtst6 libxfixes3 libxdamage1 libgtkglext1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgcc1 is already the newest version (1:8.3.0-6).
libstdc++6 is already the newest version (8.3.0-6).
libglib2.0-0 is already the newest version (2.58.3-2+deb10u2).
libc6 is already the newest version (2.28-10).
libgtkglext1 is already the newest version (1.2.0-9).
libx11-6 is already the newest version (2:1.6.7-1).
libx11-xcb1 is already the newest version (2:1.6.7-1).
libxcb-shm0 is already the newest version (1.13.1-2).
libxcb1 is already the newest version (1.13.1-2).
libxdamage1 is already the newest version (1:1.1.4-3+b3).
libxdamage1 set to manually installed.
libxfixes3 is already the newest version (1:5.0.3-1).
libxfixes3 set to manually installed.
libxrandr2 is already the newest version (2:1.5.1-1).
libxtst6 is already the newest version (2:1.2.3-1).
libpango1.0-0 is already the newest version (1.42.4-8~deb10u1).
libcairo2 is already the newest version (1.16.0-4+rpt1).
libgtk2.0-0 is already the newest version (2.24.32-3+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

and yet:

$ sudo dpkg -i anydesk_5.5.5-1_armhf.deb 
Selecting previously unselected package anydesk:armhf.
(Reading database ... 93161 files and directories currently installed.)
Preparing to unpack anydesk_5.5.5-1_armhf.deb ...
Failed to stop anydesk.service: Unit anydesk.service not loaded.
Unpacking anydesk:armhf (5.5.5) ...
dpkg: dependency problems prevent configuration of anydesk:armhf:
 anydesk:armhf depends on libglib2.0-0 (>= 2.16.0).
 anydesk:armhf depends on libgtk2.0-0 (>= 2.20.1).
 anydesk:armhf depends on libstdc++6 (>= 4.1.1).
 anydesk:armhf depends on libx11-6.
 anydesk:armhf depends on libxcb-shm0.
 anydesk:armhf depends on libxcb1.
 anydesk:armhf depends on libpango1.0-0.
 anydesk:armhf depends on libcairo2.
 anydesk:armhf depends on libxrandr2 (>= 1.3).
 anydesk:armhf depends on libx11-xcb1.
 anydesk:armhf depends on libxtst6.
 anydesk:armhf depends on libxfixes3.
 anydesk:armhf depends on libxdamage1.
 anydesk:armhf depends on libxkbfile1.
 anydesk:armhf depends on libegl1-mesa.
 anydesk:armhf depends on libgl1.

dpkg: error processing package anydesk:armhf (--install):
 dependency problems - leaving unconfigured
Processing triggers for gnome-menus (3.31.4-3) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Errors were encountered while processing:
 anydesk:armhf

pelwell, 2020-08-03:
The `armhf` in `anydesk_5.5.5-1_armhf.deb` indicates a 32-bit build. If it has dependencies on external 32-bit libraries (the 64-bit kernel shouldn't be a problem because it can host 32- and 64-bit binaries) then they won't be present on the 64-bit OS - that's the whole point.

You'll need to persuade Anydesk to host an AArch64 build.

nickpar, 2020-08-03:
i am a paid customer so i will open a ticket

thanks

nickpar, 2020-08-03:
by the way Is it my own idea or 64 bit OS is way faster ? 

i am not using the same MicroSD and i may be biased, but it loads almost instantly compared to 32 bit counterpart

melvinmajor, 2020-10-11:
Hi,
Personnally, I could install v6.0.1 on Raspberry Pi 4 2GB running latest Raspberry OS in x86 (some things aren't working great on x64 variant when I state to boot with x64 kernel so I keep the x86 for the moment).
The only problem is that I can't open AnyDesk main window nor see it on local network even though AnyDesk logo is present on status bar (next to connectivity, sound and hour).

Any idea of the problem and how to fix it?
VNC is working great and TeamViewer also but as I'm a paid user of AnyDesk, I prefer to use AnyDesk in stead of the two others.

Kind regards,

Melvin.

lurch, 2020-10-11:
> but as I'm a paid user of AnyDesk...

Then surely the onus is on AnyDesk to provide an AArch64 build, as @pelwell said earlier?

nickpar, 2020-10-12:
I have Anydesk response on this, dated August 4th 2020

"Currently, we only support Raspbian (32-bit) on Raspberry Pi 3 and newer:
https://support.anydesk.com/Installation

However, we are aware that such an AnyDesk version is requested, but we cannot give you any further information about this yet, unfortunately."

melvinmajor, 2020-10-12:
> I have Anydesk response on this, dated August 4th 2020
>
> "Currently, we only support Raspbian (32-bit) on Raspberry Pi 3 and newer:
> https://support.anydesk.com/Installation
>
> However, we are aware [...]

I'm on Raspberry OS (32-bit), I have nothing working from AnyDesk, only the presence of the icon next to the volume and network icon...

pelwell, 2020-10-12:
> I'm on Raspberry OS (32-bit), 

In which case you are in the wrong place.

nickpar, 2020-10-12:
Moreover, i have noticed practically zero issues with 32-bit Anydesk so far

melvinmajor, 2020-10-12:
> > I'm on Raspberry OS (32-bit),
> 
> In which case you are in the wrong place.

Didn't find the 32-bit section of AnyDesk to discuss issues so far when Googling it, thanks for your understanding...

melvinmajor, 2020-10-12:
> Moreover, i have noticed practically zero issues with 32-bit Anydesk so far

Okay, so I will need to reach their support today and see how to fix that.
On 64-bit, AnyDesk isn't working at all, no icon present next to the volume and network icon.
On 32-bit, you already know : present but not working at all.

Note: here's the process I use to switch on 32-bit or 64-bit kernel
In /boot/config.txt -> `arm_64bit=1 # Default value: 0 (32bit kernel launched)`

badjano, 2020-11-15:
I´m on a raspi3b with Raspberry Pi Os 64 bit, couldn´t install, says can´t install libgtk2.0:armhf and when I try to install it, it complains about another one, and this goes on recursively till I got tired

DragonSerW, 2021-01-27:
Any ideas about how to install AnyDesk @ http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-aarch64-latest.tar.gz or Raspberry Pi OS 64-bit?

badjano, 2021-01-27:
> Any ideas about how to install AnyDesk @ http://os.archlinuxarm.org/os/ArchLinuxARM-rpi-aarch64-latest.tar.gz or Raspberry Pi OS 64-bit?

You can´t, anydesk only supports 32bit arm processors for now, I´ve contacted them on reddit and that is the response I got.
I am using XRDP, it´s pretty great actually

tefracky, 2021-11-17:
Is there anything new on it? Can Anydesk be installed on RPi OS 64 bit?

Schbatzafresser, 2022-03-24:
Thank you, we had a nice time AnyDesk, but TeamViewer is now available in 64bit as deb package!