raspberrypi/Raspberry-Pi-OS-64bit #83: libreoffice fails to install

5chi-3cun-4fen, 2020-08-03:
I am now using pi4 with 8g memory. as a chinese user i changed the software source to one of the chinese ones cuz it's almost impossible to install any program with the original source. It seemed to be okay to install all the other softwares until i typed in "sudo apt install libroffice". The installation froze at about 85 percent and then it just failed. I changed the source to another one but the result turned out to be the same. I ll post the source i used later then. I ll be super glad if i can use libreoffice on the raspberry pi. Help me, por favor.

5chi-3cun-4fen, 2020-08-03:
deb http://mirrors.163.com/debian/ buster main non-free contrib
deb http://mirrors.163.com/debian/ buster-updates main non-free contrib
deb http://mirrors.163.com/debian/ buster-backports main non-free contrib
deb http://mirrors.163.com/debian-security/ buster/updates main non-free contrib

deb-src http://mirrors.163.com/debian/ buster main non-free contrib
deb-src http://mirrors.163.com/debian/ buster-updates main non-free contrib
deb-src http://mirrors.163.com/debian/ buster-backports main non-free contrib
deb-src http://mirrors.163.com/debian-security/ buster/updates main non-free contrib

5chi-3cun-4fen, 2020-08-03:
deb https://mirrors.ustc.edu.cn/debian/ buster main contrib non-free
deb https://mirrors.ustc.edu.cn/debian/ buster-updates main contrib non-free
deb https://mirrors.ustc.edu.cn/debian/ buster-backports main contrib non-free
deb https://mirrors.ustc.edu.cn/debian-security/ buster/updates main contrib non-free

deb-src https://mirrors.ustc.edu.cn/debian/ buster main contrib non-free
deb-src https://mirrors.ustc.edu.cn/debian/ buster-updates main contrib non-free
deb-src https://mirrors.ustc.edu.cn/debian/ buster-backports main contrib non-free
deb-src https://mirrors.ustc.edu.cn/debian-security/ buster/updates main contrib non-free

5chi-3cun-4fen, 2020-08-03:
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ buster main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ buster-updates main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian/ buster-backports main contrib non-free
deb https://mirrors.tuna.tsinghua.edu.cn/debian-security/ buster/updates main contrib non-free

deb-src https://mirrors.tuna.tsinghua.edu.cn/debian/ buster main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian/ buster-updates main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian/ buster-backports main contrib non-free
deb-src https://mirrors.tuna.tsinghua.edu.cn/debian-security/ buster/updates main contrib non-free

5chi-3cun-4fen, 2020-08-03:
btw, i tried to comment the "back-backports" lines. Nevertheless, it helps little.

lurch, 2020-08-03:
It's probably unlikely that those 3rd-party Debian mirrors have all the customised packages that the official Raspberry Pi repositories have? :man_shrugging:

AaronDewes, 2020-08-07:
> It's probably unlikely that those 3rd-party Debian mirrors have all the customised packages that the official Raspberry Pi repositories have? :man_shrugging:

Debian itself also doesn't have them: https://github.com/RPi-Distro/pi-gen/blob/arm64/stage0/00-configure-apt/files/sources.list
ARM64 uses debian.

AaronDewes, 2020-08-07:
@5chi-3cun-4fen What did it show before failing?
What happens if you run `sudo apt install libroffice`?

lurch, 2020-08-07:
@AaronDewes My apologies, I didn't realise that the Raspberry Pi repositories were in a [separate file](https://github.com/RPi-Distro/pi-gen/blob/arm64/stage0/00-configure-apt/files/raspi.list) and not in `sources.list` .

5chi-3cun-4fen, 2020-08-08:
> @5chi-3cun-4fen What did it show before failing?
> What happens if you run `sudo apt install libroffice`?
> 
> > I changed the source to another one but the result turned out to be the same
> 
> For that change to take effect, you need to run `sudo apt update`. Did you do that?

yes, i did run 'sudo apt update'
it seems to be the same as normal.
the output was too long so i only screenshot part of it
[image]

5chi-3cun-4fen, 2020-08-08:
[image]
[image]

AaronDewes, 2020-08-09:
It seems like all Raspberry Pi specific packages are blocked because they are in a separate repo which isn't available in China.
I don't know how the Chinese internet works, but your downloads are redirected to 117.128.6.35/cache, which could also be some company proxy.
Can you check if http://archive.raspberrypi.org/debian is accessible and are you able to download http://archive.raspberrypi.org/debian/pool/main/libr/libreoffice/libreoffice-common_6.1.5-3+rpi1+deb10u6+rpt1_all.deb with a web browser?
If not, run this command (This makes you unable to receive updates for RPi specific packages which aren't in debian and removes all rpi specific changes to packages):
```bash
sudo apt clear
sudo rm /etc/apt/sources.list.d/raspi.list
sudo apt update
sudo apt install libroffice
```

AaronDewes, 2020-08-09:
I don't know if it's possible to make a website accessible in China or get a mirror there, but if it's possible someone at @raspberrypi has to do that, and I don't know if they want to do that, because I assume it's a lot of effort.

AaronDewes, 2020-08-09:
I've now created a script which I could use to create a small mirror with GitHub actions, but I'm not sure if that's legal, and I don't want to get in trouble.

5chi-3cun-4fen, 2020-08-16:
Thank you all the same. I tried to install libreoffice today as usual and i managed to install it for some unkown reason hahaha

closed 2020-08-16 by 5chi-3cun-4fen