raspberrypi/Raspberry-Pi-OS-64bit #85: Boot Splash Screen

chester23491, 2020-08-14:
Hi Folks

We are trying to set up a boot splash screen until the user workspace is loaded successfully. 
We created this splash screen service:

```
[Unit]
Description=Splash screen
DefaultDependencies=no
After=local-fs.target

[Service]
ExecStart= /usr/bin/fbi -d /dev/fb0 --noverbose -a /home/pi/.splashscreen/splash.png
StandardInput=tty
StandardOutput=tty

[Install]
WantedBy=sysinit.target

```

After a short time, our service is interrupted and the splash screen is disappearing. Can you please help us with this issue?

Thank you!
Regards
Charles

lurch, 2020-08-14:
This question might be more suited to the https://www.raspberrypi.org/forums/ ?

chester23491, 2020-08-14:
> This question might be more suited to the https://www.raspberrypi.org/forums/ ?

Hey @lurch 
Thank you for your answer. We are using the 64 bit beta version of Rasbian, where this issue is present. Regarding this solution is our approach working on the 32 bit version:
https://raspberrypi.stackexchange.com/questions/100371/raspbian-buster-lite-splash-screen-instead-of-boot-messages-on-pi-3-model-b-a02

Something is different in the beta build - so i assumed getting help in this channel...

closed 2020-09-02 by chester23491