raspberrypi/Raspberry-Pi-OS-64bit #86: vncserver won't start after latest OS upgrade

ZengLawrence, 2020-08-16:
After upgrading Pi OS over the weekend, the vncserver couldn't start.  See below errors in daemon log:

Aug 15 14:13:34 raspberrypi vncserver-x11-serviced[2321]: /usr/bin/vncserver-x11-core: error while loading shared libraries: libmmal.so.0: cannot open shared object file: No such file or directory
Aug 15 14:13:34 raspberrypi vncserver-x11-serviced[2321]: vncserver-x11-serviced: Server exited with status 127, restarting in 5s

GHE-HS, 2020-08-17:
Same error message here:
```
pi@raspberrypi:~ $ vncserver
vncserver: error while loading shared libraries: libmmal.so.0: cannot open shared object file: No such file or directory
```
I've tried to remove the packages and reinstall them, without success!
Thanks in advance if you have any solution. In the meantime I'll try to use another VNC server.

XECDesign, 2020-08-17:
Seems to be the result of this commit - https://github.com/raspberrypi/userland/commit/f97b1af1b3e653f9da2c1a3643479bfd469e3b74

XECDesign, 2020-08-20:
@alexdelli You can monitor commits going to userland here - https://github.com/raspberrypi/userland/commits/master

GHE-HS, 2020-08-20:
@alexdelli I don't know if that question was for me but nothing has changed, I still have the same error message (after updating/upgrading).

alexdelli, 2020-08-20:
> @alexdelli You can monitor commits going to userland here - https://github.com/raspberrypi/userland/commits/master

ah ok, sorry but I'm a little noob with commit on github :)

popcornmix, 2020-08-20:
mmal is not currently supported on 64-bit. While without the revert mmal may somewhat work, it wasn't reliable (and made the 32-bit mmal unreliable).

lurch, 2020-08-20:
So I guess that means the "proper fix" is for the 64-bit `vncserver` to **not** be built against `libmmal` ?

leonardoassis-br, 2020-08-21:
Hi. I resolved this problem installing x11vnc. Tutorial: https://tinyurl.com/y4n34apb

GHE-HS, 2020-08-21:
@leonardoassis-br the nice thing with realvnc is "VNC connect" and I don't think (from what I understood) other VNC client/servers have the same functionnality (including the one you propose).

But anyway, thany you for your input, it might help other persons.

kljzndx, 2020-09-24:
vnc-server latest version is already fixed the problem

ZengLawrence, 2020-09-26:
Issue resolved after latest update.

closed 2020-09-26 by ZengLawrence