raspberrypi/Raspberry-Pi-OS-64bit #87: error while loading shared libraries: libstdc++.so.6

jonmiller1, 2020-08-17:
I'm running a program that uses opencv 4.4.  I've got it opencv 4.4 compiled and installed but the program that works under 32-bit raspbian gives me this error on 64.  I think I need to install lib32stdc++6 but it isn't in the apt-get repositories.  Any ideas?

error while loading shared libraries: libstdc++.so.6: cannot open shared object file: No such file or directory

lurch, 2020-08-17:
Are you able to do a native 64-bit build of your OpenCV application?

rag2, 2020-08-20:
You may be able to get this to work using *multiarch*:

>
>       32 bit compatibility
>
> It is possible to run *32 bit **armhf* binaries, such as 
> *W**X**to**I**mg*^1 <#sdfootnote1sym>:
>
> Debian's *multiarch*system^2 <#sdfootnote2sym> provides asolution^3 
> <#sdfootnote3sym> by allowing an *armhf* capability to be added, as an 
> additional supported architecture.
>
> sudo dpkg --add-architecture armhf
>
> To resolve an error around any package, usually a library package, 
> requires installing the *armhf*version of that package, denoted by an 
> *:armhf*suffix. If *libX11.so.6*, for example, is required:
>
> sudo apt install libx11-6:armhf
>
> *W**X**to**I**mg*, for example, requires2 further *armhf* libraries:
>
> sudo apt-get install libxft2:armhf
> sudo apt-get install libasound2:armhf
>
> 1 <#sdfootnote1anc>This program is 
> *commercialware*→*abandonware*→*freeware*, with *no source* available. 
> See https://wxtoimgrestored.xyz/beta/.
>
> 2 <#sdfootnote2anc>See https://wiki.debian.org/Multiarch/HOWTO.
>
> 3 <#sdfootnote3anc>Adapted from a description at 
> https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues/62#issuecomment-647459572.
>
Effectively that allows both 64 bit (*arm64*) and 32 bit (*armhf*) 
libraries to co-exist independently on the same machine.

Robin

On 17/08/2020 21:59, jonmiller1 wrote:
> I'm running a program that uses opencv 4.4. I've got it opencv 4.4
> compiled and installed but the program that works under 32-bit
> raspbian gives me this error on 64. I think I need to install [...]

lurch, 2021-06-07:
More info about this error (and a workaround) in #156