raspberrypi/Raspberry-Pi-OS-64bit #89: Problems with booting on 4k monitors

MikeDB1, 2020-08-19:
I have a 32" Samsung 4k monitor which Pis always give very small text on so in the 32 bit system I use raspi-config to set a 1920x1060 resolution so as to get larger text size. But on the 64 bit system I've tried all sorts of settings but as far as I can make out the screen resolution in raspi-config is being ignored or misinterpreted as the text size never changes.

Connecting an old 1440x? 19" HDMI monitor works fine so the automatic detection does work, but if I use automatic detection on the Samsung the display flashes on and off randomly as well as having very small text.

I reinserted the 32 bit memory card just to check and everything still works fine on that.

Also tried it with a 42" Toshiba TV and a reflashed memory card and after flashing up the 64 bit boot screen message the HDMI out goes black and eventually turns off. Replugging into the old monitor and the system starts to initialise correctly.

lurch, 2020-08-19:
Some of these may help?  (I don't have a 4K monitor myself)
https://www.raspberrypi.org/documentation/configuration/arandr.md
https://www.raspberrypi.org/documentation/configuration/hdmi-config.md
https://www.raspberrypi.org/documentation/configuration/config-txt/video.md
https://www.raspberrypi.org/blog/raspbian-update-screen-sizes/

MikeDB1, 2020-08-20:
Thanks.  I went to give those a try but at the moment the monitor is flashing on for under a second then off for 30 seconds.  I'll try it again tonight when hopefully it's in a better mood :-)

lurch, 2020-08-20:
Perhaps the EDID in the monitor is reporting a mode that the Pi can't output? (And/or the Pi is outputting a mode that your monitor can't display?)
If you have some other form of connection to the Pi (e.g. SSH or UART) with which you can debug things, it might be worth opening an issue at https://github.com/raspberrypi/firmware  (although if the 32-bit image works fine, and it's only the 64-bit image that results in a blank display, opening an issue at https://github.com/raspberrypi/linux may be more appropriate).
Ping also @6by9

6by9, 2020-08-20:
Are you booting into X? If so then the Screen Configuration app is the correct place to set the resolution. If you can't see the screen well enough to be able to use that, then SSH in and use ```DISPLAY=:0 xrandr --output HDMI-1 --mode 1920x1080```. Please then use the Screen Configuration tool to set the mode again - doing it from the command line does not persist over a reboot.

MikeDB1, 2020-08-20:
By "Screen Configuration app" do you mean raspi-config or something else ?   Setting raspi-config is definitely being ignored.

As for the EDID, surely to be HDMI compliant a product has to recognise and obey all EDIDs, or report that it cannot ?    It's a standard Samsung monitor, not some dodgy Chinese brand with a fake ID.

pelwell, 2020-08-20:
I think @6by9 means the applet you find at Raspberry -> Preferences -> Screen Configuration...

6by9, 2020-08-20:
I do mean that app. It's ```arandr``` under the bonnet.

6by9, 2020-08-20:
TBH I can't say at present why your monitor isn't driven properly in the 64bit OS but is fine with 32bit.

64bit is still a beta, and there are a load of other changes to come, therefore I'm not fretting over this at present.

X does ignore any setting that you add to the kernel command line - there's no easy way around that. That's why the Screen Configuration app is there. It also allows you to configure how multiple monitors are arranged - something config.txt is never going to convey.

MikeDB1, 2020-08-20:
Okay I've dug out some other monitors to check things over so I'll summarize the results.

Firstly the Screen Configuration app does what it says on the tin even when raspi-config or the other methods mentioned fail so I'll stick with this.

Screens up to 2560W work reliably with the 64 bit OS (and the 32 bit OS), and I can set them into lower resolution modes successfully.

Two 4k screens (Samsung U32J590 4K Ultra HD 32" LED Monitor and a Toshiba 42" LED TV)  are always fine on the 32 bit OS, but for some reason the 64 bit OS doesn't like them when booting in 4k mode - I either get no display or them flickering on and off every few seconds.

If forced to a lower resolution using the Screen Configuration tool then they seem to be fine on the 64bit OS and remember this setting on reboot.

However I also notice that when hotplugging the monitors they aren't acting on the information from the newly plugged in monitor but are just retaining the information from the monitor plugged in at boot.  I believe the HDMI spec allows for hot-plugging so this is also incorrect operation.

I haven't dare try doing an apt-update yet to see if that messes things up.  

So in conclusion I now have 4k monitors working in lower res modes so can continue testing the 64 bit OS, but it's obviously got a few bugs with full 4k mode that need addressing as they work fine in the 32 bit OS.

6by9, 2020-08-20:
Hotplug has never been supported on the Pi. It may be when we switch to the full KMS driver, but that won't be for a while.

Please provide the output from ```vcgencmd version``` and ```uname -a``` on both the working 32bit and non-working 64bit systems. Almost all of the display driver code is common between the two builds, so they should both work fine.

lurch, 2020-08-20:
To get a better idea of why things work differently between 32-bit and 64-bit, and between 64-bit console mode and 64-bit X11 mode, have a read through some of https://github.com/raspberrypi/documentation/search?q=fkms&type=Code

MikeDB1, 2020-08-20:
Okay.  I thought Hot-plug was a requirement of HDMI 1.3 though, not an option.  

I'll have to go back to the 32 bit version to get the outputs, but the 64 bit one is what you provide as an update to a freshly zapped memory card when first booting.

vcgencmd : August 6th 16:22:25 then some UUID too long to retype.

uname : 5.4.51 -v8+  #1333 Aug 10th 16:58:35

As you say, almost all of the code is common but there's obviously a small difference.