raspberrypi/Raspberry-Pi-OS-64bit #91: PING returns inconistent results

kurt-lefevre, 2020-08-24:
I've a 4B with the up to date version of the 64 bit OS as found on http://downloads.raspberrypi.org/raspios_arm64/images/

Im running a bash script pinging every 5 seconds a device on my local network. Ad-hoc ping returns error codes -1 and -2 where it should be 0 as the pinged device is reachable. 
The PI is wired on the same router as the pinged device.

This script worked for years on a B+ running 32 bit Jessy and Buster. I recently replaced the B+ with a 4B which was the only change I made.

ghollingworth, 2020-08-24:
From 'man ping'

There are no negative error codes...  it should return 0 on success, 1 (if it failed to receive any ping but otherwise everything is fine), or 2 (general other failure)

Are you just getting destination host unreachable?

Gordon

kurt-lefevre, 2020-08-24:
Yes, sure, no neg return codes. Typo from my side.

You can find the script below. The PI is used as an DLNA music server running Minim. 
The script runs in the background and is used on a headless install. The most interesting functions are loop() and checkStreamer().

I initially checked only on 0 (success) in the loop(), but then I figured out I also got 2 values (which is strange). Randomly I also get 1 values, which actually causes a shutdown. This is as defined, but ping shouldn't return 1 or 2 as the device is there (a streamer is playing).

-----------------------

#!/bin/bash
# script started automatically via /etc/rc.local
 
# Configuration parameters
IP_ND5XS="192.168.1.205"
IP_MUSO="192.168.1.207"
LOGFILE="/home/pi/dlna-monitor.log"
MAX_LOGFILE_SIZE=10240
CHECKFILE="/boot/nodlna"
SLEEPTIME=5
SLEEP_AT_START=120

# Do not modify the script variables below
RESULT=0

# Functions
function checkStreamer () {
	ping -c1 -s1 -W1 $1 > /dev/null
	RESULT=$?
	if [ $RESULT -eq 2 ]; then
		echo $(date) - Ping failed for $1 - Result: $RESULT >> $LOGFILE
	fi
}

function checkFileSize () {
	FILESIZE=$(stat -c%s $LOGFILE)
	if [ $FILESIZE -gt $MAX_LOGFILE_SIZE ]; then
		rm $LOGFILE > /dev/null
		echo ----------------------------------- START ------------------------------- >> $LOGFILE
		echo $(date) - Deleted $LOGFILE >> $LOGFILE
	fi
}

function loop () {
	# Keep looping as long as ND5XS and/or MUSO is on
	while [ $RESULT -ne 1 ]; do
		sleep $SLEEPTIME
		checkStreamer $IP_ND5XS
		if [ $RESULT -eq 1 ]; then
			checkStreamer $IP_MUSO
		fi
	done
}

function checkBypass() {
	# Check fo bypass file. If found, stop here to avoid an unwanted shutdown
	if [ -f $CHECKFILE ]; then
		# bypass file found. Stop process here
		echo $(date) - Bypass file found >> $LOGFILE
		echo ----------------------------------- STOP -------------------------------- >> $LOGFILE
		exit 0
	else
		echo $(date) - No bypass file found >> $LOGFILE
	fi
}

########################################### SCRIPT START #####################################
# Make sure sources are up, or allow time to touch $CHECKFILE or kill process in case of problems
echo ----------------------------------- START ------------------------------- >> $LOGFILE
sleep $SLEEP_AT_START

# Check logfile size
checkFileSize

# Check for bypass file in /boot
checkBypass 

# Keep looping till sources are swithced off
loop

# Both ND5XS and Muso are considered to be off -> Shutdown PI
echo $(date) - Music sources are switched off >> $LOGFILE
echo ----------------------------------- STOP -------------------------------- >> $LOGFILE
sudo poweroff

ghollingworth, 2020-08-24:
I would guess you are seeing the network go down occasionally due to DHCP leases timing out?

Otherwise it's possible there's just a short drop while the link is re-negotiated at 1GHz...  I'm not sure if this is a thing or not, but you're not retrying on failure so it's possible this is the case....

Can you just run ping constantly and pipe to a file, then when it drops out have a look at the file and see how long it dropped out for.  It's possible it's only a short period, so a retry would fix it.

I don't  think this is anything to do with 64bit though, I'm sure this will be exactly the same on the 32 bit OS

kurt-lefevre, 2020-08-24:
- No, I haven't seen any network dropouts when this occurs.

- I don't retry as the goal is to shutdown the PI when my 2 sources are swithed off. But that's a good point, I'll add a retry.

- I'll also ping to a file. The thing is that it's really random. Yesterday I listend music for 14h without any anomalies in the log. This morning it got a 1 return twice...

ghollingworth, 2020-08-24:
From one of our engineers:

`sudo mii-tool eth0 -v`

Tells you the link state and advertised capabilities

`sudo mii-tool eth0 -w`

Puts it in polling mode and reports link events.

kurt-lefevre, 2020-08-24:
I both started the ping to a file and the mii-tool. As soon as I've something I'll let you know.
The annoying about it is that it's unpredictable when it happens.

kurt-lefevre, 2020-08-24:
The mii tool returned this:
```
17:15:48 eth0: negotiated 1000baseT-FD flow-control, link ok
17:15:53 eth0: no link
17:15:56 eth0: negotiated 1000baseT-FD flow-control, link ok
```
This was just after a reboot. What do I learn from this?

ghollingworth, 2020-08-24:
This suggests that your link is dropping out occasionally, which is likely due to a poor cable or the switch dropping the link for other reasons.  You could try forcing it to 100baseT and see if it still happens, or try a short Ethernet cable to see if you get the same response

kurt-lefevre, 2020-08-24:
* The cables are recent Telegartner cat7 cables, so I don't question those right now. The one from the PI to the router is 0,5m; the one from the router to the streamer is 5m. Shouldn't be in the danger zone.
* I'll hard reboot the switch, you never know.
* On the B+, it was indeed 100baseT. How do I force this?

Tx,
k

P33M, 2020-08-25:
You can force 100baseTx with 
`sudo mii-tool -A 100baseTx-FD eth0`
The setting will not persist across reboots.

kurt-lefevre, 2020-08-25:
I updated the PI with the latest available updates yesterday evening.
Today, using a 1GB connection, I haven't got any "no link" events since a 10h uptime.
I'll keep an eye on it.

kurt-lefevre, 2020-08-31:
It turns out that under heavy network load ping doens't return in time. I also get this when running the script on another linux box pinging other devices as well. So, I guess it's because the wireless router is pushed to its limits.

kurt-lefevre, 2020-09-12:
Some more investigation has been done:
* set speed to 100 mb/s -> Still line drops
* used static ip addresses instead of DHCP (address reservation)) -> Still line drops

**However, and this may require the OS folks their attention, the line drops always after a PI uptime of 15'! Thereafter, the connection remains stable. This looks strange to me and makes me thinking about a driver issue.**

Tomorrow I'll try with a new gigabit switch.

Keep you posted on my findings.

ghollingworth, 2020-09-14:
Have you tested this on the 32bit version ?  Are you able to reproduce it there?

kurt-lefevre, 2020-09-14:
I haven't tried the 32 bit version as I can't benefit from the 8G RAM in this case.
However, I ran it since years on a B+ in 32 bit without any issus. Of course, this is different hardware.

But... Since yesterday evening I've added a new switch to the router and connected both the PI 4B and my streamer on it. No link drops anymore since then. Might be an issue with the switch in the router. I'll follow up on it.

closed 2020-09-14 by ghollingworth