raspberrypi/Raspberry-Pi-OS-64bit #92: USB Wireless keyboard/mouse not recognized after reboot

ZengLawrence, 2020-08-27:
Kernel failed to recognize my Microsoft Sculpt keyboard and mouse, which use a USB wireless dongle.  I have to unplug and re-plug in to get it recognize.  It works after re-plug it.  No issue with 32-bit OS, which recognize after each reboot.  

I have another USB mic input, which works with every reboot.

Here is dmesg log:
[   33.326234] usb 1-1.4: device descriptor read/64, error -110
[   33.513945] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[   38.098767] usbcore: registered new interface driver snd-usb-audio
[   38.725907] usb 1-1.4: new full-speed USB device number 6 using xhci_hcd
[   38.834981] usb 1-1.4: New USB device found, idVendor=045e, idProduct=07a5, bcdDevice= 7.97
[   38.834990] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   38.834995] usb 1-1.4: Product: Microsoft® 2.4GHz Transceiver v9.0
[   38.835000] usb 1-1.4: Manufacturer: Microsoft
[   38.842939] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:045E:07A5.0002/input/input1
[   38.902294] hid-generic 0003:045E:07A5.0002: input,hidraw1: USB HID v1.11 Keyboard [Microsoft Microsoft® 2.4GHz Transceiver v9.0] on usb-0000:01:00.0-1.4/input0
[   38.912825] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.1/0003:045E:07A5.0003/input/input2
[   38.913061] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.1/0003:045E:07A5.0003/input/input3
[   38.913213] hid-generic 0003:045E:07A5.0003: input,hidraw2: USB HID v1.11 Mouse [Microsoft Microsoft® 2.4GHz Transceiver v9.0] on usb-0000:01:00.0-1.4/input1
[   38.922804] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.2/0003:045E:07A5.0004/input/input4
[   38.982085] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 System Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.2/0003:045E:07A5.0004/input/input5
[   38.982242] hid-generic 0003:045E:07A5.0004: input,hidraw3: USB HID v1.11 Device [Microsoft Microsoft® 2.4GHz Transceiver v9.0] on usb-0000:01:00.0-1.4/input2

pelwell, 2020-08-27:
Duplicate of https://github.com/raspberrypi/linux/issues/3334 - there is a fix in the latest bootloader.

ZengLawrence, 2020-08-27:
Thanks, @pelwell.  After upgrading to latest stable bootloader, it resolved this problem.

closed 2020-08-27 by ZengLawrence