raspberrypi/Raspberry-Pi-OS-64bit #96: Boot from USB 1000mS timeout is too short

MikeDB1, 2020-09-07:
I'm trying to use an individually switched USB hub to boot 64bit OS from a USB Flash stick so that I can compile on the Pi4 and download the output to another flash stick, then power down and switch off the OS Flash so that the bare metal flash stick will be used.   But the USB hub adds a delay which is too long for the timeouts on the boot sequence so I get a SD card not found message.  I can't see anywhere to extend the 1000mS timeout stated - 'wait for network' doesn't work for a USB stick.

andrum99, 2020-09-16:
I suggest you post your query on the troubleshooting forum at https://www.raspberrypi.org/forums/viewforum.php?f=28&sid=a50a74ffb1a74ba290881b490ce875dd. This repo is for tracking bugs specific to the 64-bit version of Raspberry Pi OS. It sounds like the problem you are having will occur on other OSes as well.

MikeDB1, 2020-09-16:
No it works fine for 32bit PiOS (or whatever it's called this week).  The 64bit PiOS appears to be slower and so needs a longer default timeout.

pelwell, 2020-09-16:
That doesn't make sense - the Pi 4 EEPROM loads the same firmware regardless of the OS; it doesn't even know you'll be booting a 64-bit OS.

andrum99, 2020-09-16:
It could be a subtle timing issue - the 64-bit kernel is uncompressed so could take slightly longer to load from the boot filesystem. 

Anyway, there seems to be a dialog between Pi and the OP now, which was primarily what I was aiming for with my suggestion.

MikeDB1, 2020-09-16:
Fair point.  I've eight Pi4s here and some may not have the latest EEPROM which could explain the difference. However the problem still holds - can we have a longer timeout please.

pelwell, 2020-09-16:
Are you not talking about the `USB_MSD_LUN_TIMEOUT` configuration setting, as described here?:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2711_bootloader_config.md

pelwell, 2020-09-16:
(or one of those)

MikeDB1, 2020-09-16:
Well according to this document the default is 20 seconds, which it isn't.  But let me try again tomorrow and play with some of these values to see if I can reliably boot after playing with the values.

pelwell, 2020-09-16:
If you find that there is an inadequate timeout, especially one that can't be modified, then I'm sure we'll be able to add it - that's one of the nice things about having the boot loader in EEPROM.

lurch, 2020-09-16:
...and if you haven't done so already, it may be worth experimenting both with and without the USB hub, to see if that has any impact on things.

MikeDB1, 2020-09-16:
> ...and if you haven't done so already, it may be worth experimenting both with and without the USB hub, to see if that has any impact on things.

The whole point is I need the switched hub so as to switch between the compiler Pi4 and the target bare metal one without constantly plugging.  It works fine without the hub but that's irrelevant

lurch, 2020-09-17:
> It works fine without the hub but that's irrelevant

It **is** relevant, because it shows that regular USB boot _does_ work from your USB Flash stick, and it's "only" the USB hub that is causing things to fail. Thanks for the clarification :+1: 
Do you have more details (e.g. a product link) about the particular hub you're using? Are you able to test with a regular USB hub (one without individual switches) to see if that also causes problems?

MikeDB1, 2020-09-17:
I didn't say USB boot doesn't work - I said it can time out because some USB hubs have a longer initialisation time than the time the Pi allows.  An unswitched hub works.   In any case I'll try playing with the values in the document pelwell refered me to.

lurch, 2020-09-17:
> An unswitched hub works.

Thanks, that's also useful confirmation.
I wonder if it might be a "noise" issue rather than a "timeout" issue? If you e.g. boot your Pi from an SD card, do you get different read-write speeds to your USB flash stick when plugged in directly / plugged in via an unswitched hub / plugged in via your switchable hub?

MikeDB1, 2020-09-17:
Can't find the exact USB switch I've got as it seems to be discontinued but it's a USB2.0 version of this :

https://smile.amazon.co.uk/UGREEN-Sharing-Keyboard-Scanner-Printer-Black/dp/B01N6GD9JO/ref=smi_www_rco2_go_smi_1285150262?_encoding=UTF8&hvadid=309968313994&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9046436&hvnetw=g&hvpone=&hvpos=&hvptwo=&hvqmt=&hvrand=6784479521780918645&hvtargid=pla-350939791158&ie=UTF8&linkCode=df0&psc=1&tag=googshopuk-21

There's hundreds of similar Chinese devices like this.

lurch, 2020-09-17:
That product you linked to seems to be for switching all connected USB devices between two different USB host devices, rather than for selecting which USB devices are visible to a single USB host?

MikeDB1, 2020-09-17:
Yes it is.  The 2.0 one I have has two USB host leads but also has an On/Off switch for each of the USB devices.  The idea was once it worked to switch to having a cross compiling Pi and a separate bare metal Pi.  But as it doesn't work I'm now swapping USB sticks physically between two Pis.

lurch, 2020-09-17:
> But as it doesn't work I'm now swapping USB sticks physically between two Pis.

You may want to look at network booting (as that'll probably be easier than physically swapping USB sticks), although I've no idea if that works for bare-metal stuff.

Did you manage to try tweaking some of the EEPROM timeout values? Did you try connecting _both_ USB host leads to a single Pi in case it was the "other USB host port" on your hub that started up initially-active?

MikeDB1, 2020-09-17:
> You may want to look at network booting (as that'll probably be easier than physically swapping USB sticks), although I've no idea if that works for bare-metal stuff.

Good idea.  It works for bare metal on the old 3B but haven't tried for the 4 since it got latest boot software. 
  
Tried each of the host leads individually.  Don't see how both at same time will help.
Fighting other issues so haven't got back to real work yet.  Will try later on.