raspberrypi/Raspberry-Pi-OS-64bit #97: Warning when downloading package source

andrum99, 2020-09-22:
I'm getting a warning:

> W: Download is performed unsandboxed as root as file 'raspberrypi-net-mods_1.3.0.dsc' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)

Steps to reproduce:

1. Edit `/etc/apt/sources.list.d/raspi.list` and uncomment the source line.
2. `sudo apt update`.
3. `sudo apt-get source raspberrypi-net-mods`.

Full output:

> pi@pi4b-64:~ $ sudo apt-get source raspberrypi-net-mods
> Reading package lists... Done
> Need to get 3,312 B of source archives.
> Get:1 http://archive.raspberrypi.org/debian buster/main raspberrypi-net-mods 1.3.0 (dsc) [592 B]
> Get:2 http://archive.raspberrypi.org/debian buster/main raspberrypi-net-mods 1.3.0 (tar) [2,720 B]
> Fetched 3,312 B in 0s (34.9 kB/s)                
> dpkg-source: info: extracting raspberrypi-net-mods in raspberrypi-net-mods-1.3.0
> dpkg-source: info: unpacking raspberrypi-net-mods_1.3.0.tar.xz
> W: Download is performed unsandboxed as root as file 'raspberrypi-net-mods_1.3.0.dsc' couldn't be accessed by user '_apt'. - pkgAcquire::Run (13: Permission denied)

XECDesign, 2020-09-22:
No root access needed when running apt source.

andrum99, 2020-09-22:
So it's user error? My mistake, closing.

closed 2020-09-22 by andrum99