raspberrypi/Raspberry-Pi-OS-64bit #99: Can not install Google Chrome

edurenye, 2020-09-27:
I tried to install the 'Movistar+' Addon to Chromium and it did not work, then I tried to install Google Chrome from the official web site to see if I could install it there. I downloaded the 64bit deb and tried to install, but it gave an error: 'The file could not be installed' without any extra info.

I'm using a Raspbery pi 4.

rag2, 2020-09-27:
Eduard,

the file that you found is for Intel/AMD x86_64 architecture (amd64). 
There appears to be no official build of Chrome for arm64, or armhf. 
(The Movistar+ addon probably requires DRM capabilities.)

You may find Kodi, or a derivative such as OpenELEC, will let you deal 
with Movistar.

Robin

On 27/09/2020 19:02, Eduard Reñé Claramunt wrote:
> I tried to install the 'Movistar+' Addon to Chromium and it did not
> work, then I tried to install Google Chrome from the official web site
> to see if I could install it there. I downloaded the [...]

edurenye, 2020-09-28:
I expected that there might be something related to DRM missing in chromium, that is why I tried to install Google Chrome.
Regarding Kodi, there is no plugin for Movistar+, there is just one to connect the decoder, but not to use the web.
Also I found problems with Netflix plugin on kodi, there are issues filled, but not fixed.
All this stuff just works fine out of the box in Google Chrome.

That is why I filled a issue for Google Chrome, but you pointed out that there is missing build for this architecture, then I should fill an issue for google chrome.

edurenye, 2020-09-28:
Is there any way to add the DRMs that has Google Chrome to Chromium?

edurenye, 2020-09-28:
I couldn't really find where to file this feature for Google Chrome

pelwell, 2020-09-28:
Raspberry Pi is not considered a secure boot platform by the content providers, which is why there is no official DRM support.

lurch, 2020-09-28:
Should this issue be closed? AFAIK this repo is largely for tracking issues that affect the 64-bit version of Raspberry Pi OS that work in the 32-bit version of Raspberry Pi OS; but Google Chrome isn't available for either version?

edurenye, 2020-09-30:
I discovered this https://lemariva.com/blog/2020/06/raspberry-pi-amazon-prime-netflix-and-drm-solution
And seem to have support for https://gist.github.com/lemariva/0eb4ff4e847700627a5ebb71711c31bf Raspberry Pi OS 64bit
I'll test that, but any issues should go there, and I agree with @lurch we can close this issue.

closed 2020-09-30 by edurenye