raspberrypi/Raspberry-Pi-OS-64bit #100: Getting "invalid output pipe-menu /usr/bin/obamenu" headless

PtrckM, 2020-10-02:
- I'm getting "invalid output pipe-menu /usr/bin/obamenu" when connecting to rpi4 headless thru realvnc viewer
- ssh is okay

[image]

XECDesign, 2020-10-02:
How did you get there? Did you install openbox onto a lite image?

PtrckM, 2020-10-03:
Just like a normal day, just started my rpi4 every morning I thought when I connect to it via realvnc its not connecting at all or the headless aren't working. to my surprise when I right click I'm indeed on the rpi4 GUI but different environment perhaps? Cause the taskbar is gone and desktop icons too.

lurch, 2020-10-03:
ping @spl237 could this be some settings file getting corrupted?

spl237, 2020-10-03:
Possibly, but it seems pretty extreme if so. You could try a reset to defaults in Appearance Settings, but it looks to me as if something else is corrupted somewhere.

PtrckM, 2020-10-03:
[image]

- browser works, terminal emulator
- tried click restart , reconfigure nothing seems happening, i try to click desktop preferences it shows

[image]

pelwell, 2020-10-03:
I think you'll save yourself a lot of time by making a backup of your user data and starting again.

XECDesign, 2020-10-05:
`/var/log/apt/history.log` might give a hint if apt ended up removing packages for some reason.

avalon-reimu, 2021-09-29:
Can you install raspberrypi-ui-mods?