raspberrypi/Raspberry-Pi-OS-64bit #101: CDROM has not been defined or cannot be found

DagBjorndal, 2020-10-03:
After installing 'abcde', I get the folowing message when trying to activate:

[ERROR] abcde: CDROM has not been defined or cannot be found

I hope this is helpful info.

pelwell, 2020-10-04:
This is almost certainly the wrong place to report this issue - you'd be better off in the Raspberry Pi Forums, where they will almost certainly want to know if you have a CD-ROM drive attached...

DagBjorndal, 2020-10-04:
Helpful interpretation for almost certainly non-professional readers:
Repository installed 'abcde' works in the 32 bit version of Raspberry Pi OS, with the USB connected CDROM in question, but not in the 64 bit version.

pelwell, 2020-10-04:
> Repository installed 'abcde' works in the 32 bit version of Raspberry Pi OS, with the USB connected CDROM in question, but not in the 64 bit version

So there is no misunderstanding, is that a statement of fact according to your experience?

lurch, 2020-10-04:
Do you get the same `lsusb` output in both the 32-bit and 64-bit versions of Raspberry Pi OS? What version numbers of `abcde` get installed into each OS?

DagBjorndal, 2020-10-04:
Lsusb as expected. Same version abcde v2.9.3.

Turns out I have a dicey CDROM cable :-P

I can confirm that both _abcde_ and _dvdbackup_ works in Raspberry Pi OS 64 bit on my equipment. Case closed. Thanks.

closed 2020-10-04 by DagBjorndal