raspberrypi/Raspberry-Pi-OS-64bit #106: irs1125 unicam not working correctly

bjajoh, 2020-10-14:
Hi, I tried to test our depth camera on the 64Bit OS 
The aarch64 kernel module (irs1125.ko) is working fine and detection the camera (i2c), no obvious issues there.
However it is not working correctly. It is not recognized as a "V4L2_CAP_VIDEO_CAPTURE" device.
Moreover there is no unicam (/dev/video0 & /dev/video1) device under "v4l2-ctl --list-devices".
"vcgencmd get_camera" also returns "supported=1 detected=0".
It seems like something is going wrong on the csi2 <-> v4l2 side.
I tested it with the 5.4.70 kernel on a pi4 (4gb). The 32bit version is just working fine with the exact same code, it is a aarch64 related issue.

Can you point me into the right direction?

Thanks for your help!
Bjarne

6by9, 2020-10-14:
```vcgencmd get_camera``` is unrelated as the firmware knows nothing about your camera module so will never detect it.

Try enabling the debug in bcm2835-unicam (```bcm2835_unicam.debug=5``` in /boot/cmdline.txt) to see if it reports any issues. Otherwise check whether both drivers are probing and registering cleanly with the v4l2_async_notifier framework.

I've just tested imx477 on the rpi-5.9.y branch and it all works fine.
We don't have an IRS1125 module to test with.

bjajoh, 2020-10-15:
@6by9 Thanks for your quick reply! No issues with "bcm2835_unicam.debug=5".
What do I have to pay attention to regarding the v4l2_async_notifier framework?

6by9, 2020-10-15:
Just tested on a 5.9.0 arm64 kernel build with a nobbled irs1125 driver (all I2C reads succeed) and it's created /dev/video0 and /dev/video1 fine.
I'm still on a 32bit OS install, but I wouldn't expect that to make any difference.

bjajoh, 2020-10-18:
I just double checked with the 32bit OS and the 64bit kernel. Everything is working fine with the 32bit kernel, after switching to `arm_64bit=1` and rebooting it stops creating the /dev/video1. Do you have any more idea what I should check?
Thanks for you help! :)

bjajoh, 2020-10-18:
I just double checked with the 32bit OS and the 64bit kernel. Everything is working fine with the 32bit kernel, after switching to arm_64bit=1 and rebooting it stops creating the /dev/video1. Do you have any more idea what I should check?
Thanks for you help! :)

6by9, 2020-10-19:
> it stops creating the /dev/video1.

Do you get /dev/video0?
To enumerate the metadata pad the sensor subdev needs to define 2 source pads, complete with formats and sizes. That should be common between 32 and 64bit kernels though.

bjajoh, 2020-10-19:
Sorry, I was unclear. No, I'm neither getting /dev/video0 nor /dev/video1.

bjajoh, 2020-11-14:
It turned out that the unicam was simply not enabled with the kernel I tested. This PR solved it. :) https://github.com/raspberrypi/linux/pull/3890

closed 2020-11-14 by bjajoh