raspberrypi/Raspberry-Pi-OS-64bit #107: No USB on Rspberry Pi 4 4GB

axelsimon, 2020-10-15:
I tried [2020-08-20-raspios-buster-arm64](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/) on a Raspberry Pi 4, 4GB, and my USB keyboard was never recognised. Neither was my external USB 3.1 hard drive.

Tried adding the classic `max_usb_current=1` to /boot/config.txt but to no avail (the external hard drive is self-powered anyway).

Nothing shows up in `dmesg`, much less when running `lsusb`.

That's about all i could look for, there really is no sign of anything usb being detected, as far as i can tell.

pelwell, 2020-10-15:
On the 64-bit image, what does `dmesg | grep pci` show?

Does USB work with a normal 32-bit RPiOS image?

lurch, 2020-10-15:
Have you tried booting with just the keyboard, without the harddrive being plugged in as well? What model of keyboard is it? Have you tried swapping it between different USB ports?

axelsimon, 2020-10-15:
I have tried with just the keyboard, plugged in any of the two USB 2 ports. No difference.
I think i tried both USB 3 ports for the hard drive too.

It's a TypeMatrix 2030 USB Keyboard (ID 1e54:2030)

I'll try what you suggest, @pelwell.

axelsimon, 2020-10-15:
I've started the same rpi4 with a 32 bit version of Raspberry Pi OS (2020-08-20-raspios-buster-armhf-lite) and it all looks the same.
I tried my keyboard on all 4 USB ports, nothing shows up in dmesg. Nothing in lsusb either.

> pi@raspberrypi:~ $ dmesg | grep pci
[    0.948663] brcm-pcie fd500000.pcie: host bridge /scb/pcie@7d500000 ranges:
[    0.951373] brcm-pcie fd500000.pcie:   No bus range found for /scb/pcie@7d500000, using [bus 00-ff]
[    0.954100] brcm-pcie fd500000.pcie:      MEM 0x0600000000..0x0603ffffff -> 0x00f8000000
[    0.956812] brcm-pcie fd500000.pcie:   IB MEM 0x0000000000..0x00bfffffff -> 0x0000000000
[    1.550998] brcm-pcie fd500000.pcie: link down

Starting to look like a hardware issue :/

pelwell, 2020-10-15:
You could try turning up the voltage a little - `over_voltage=4` in config.txt should do it - but if that doesn't work then I agree with you.

How long have you had the 4B, and has the USB ever worked?

axelsimon, 2020-10-15:
I've had it since may i think. The USB did work i think, i tried it a few months back with a keyboard (another TypeMatrix 2030) and it was fine.

I'll try to over volt a bit, let's see.

lurch, 2020-10-15:
Probably a stupid question, but I assume you've checked that the keyboard still works okay on some other computer?
Do you have e.g. a USB flash drive that you could plug into the Pi 4B, to see if that shows up in `lsusb`?

axelsimon, 2020-10-16:
Don't worry, i'd ask the same question if i were in your place. Yes, unfortunately i've tried both keyboards on both my laptop and the rpi 4, and they both work on the laptop and are not detected on the rpi4. At this point, i'm going to assume this is a hardware fault on this specific rpi4 and get in touch with Pimoroni, from whom i bought it.

Closing this, thanks a lot for your help @lurch & @pelwell.

closed 2020-10-16 by axelsimon

min918, 2020-12-05:
i got the same problem....
lsusb nothing