raspberrypi/Raspberry-Pi-OS-64bit #110: File system is always error when using hard disk.

dishangti, 2020-10-18:
I use a hard disk connected on USB3 port where 
the system is. It works well when power is on as a web server. However, when I reboot it, it sometimes cannot start again. When run fsck on another computer, it shows some file system error, after mending them, it works again.

lurch, 2020-10-19:
Do you get the same behaviour with 32-bit Raspberry Pi OS, or is this definitely specific to only 64-bit Raspberry Pi OS?

What make/model harddrive are you using? Is it USB-powered or does it have its own power supply?

dishangti, 2020-10-21:
Sorry, I didn't try it on 32-bit OS, because I haven't got another empty disk. The model I'm using is Raspberry Pi 4B with 8GB RAM, and the power is supplied by the official power adapter, which can provide 5V 3A, also for the disk, so I think the power may not be the reason.
Today the file system got broken again, these below are the fsck's output:

fsck from util-linux 2.33.1
e2fsck 1.44.5 (15-Dec-2018)
rootfs contains a file system with errors, check forced.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Unattached inode 349
Connect to /lost+found<y>? yes
Inode 349 ref count is 2, should be 1.  Fix<y>? yes
Unattached inode 256714
Connect to /lost+found<y>? yes
Inode 256714 ref count is 2, should be 1.  Fix<y>? yes
Pass 5: Checking group summary information
Free blocks count wrong (119473865, counted=119473863).
Fix<y>? yes
Inode bitmap differences:  -61
Fix<y>? yes
Free inodes count wrong for group #0 (3668, counted=3669).
Fix<y>? yes
Free inodes count wrong (29485697, counted=29485701).
Fix<y>? yes

rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: 75899/29561600 files (0.2% non-contiguous), 2556223/122030086 blocks

I have found the file of MariaDB is broken before checking it.

dishangti, 2020-10-21:
> Do you get the same behaviour with 32-bit Raspberry Pi OS, or is this definitely specific to only 64-bit Raspberry Pi OS?
> 
> What make/model harddrive are you using? Is it USB-powered or does it have its own power supply?

Sorry, I didn't try it on 32-bit OS, because I haven't got another empty disk. The model I'm using is Raspberry Pi 4B with 8GB RAM, and the power is supplied by the official power adapter, which can provide 5V 3A, also for the disk, so I think the power may not be the reason.
Today the file system got broken again, these below are the fsck's output:

fsck from util-linux 2.33.1
e2fsck 1.44.5 (15-Dec-2018)
rootfs contains a file system with errors, check forced.
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Unattached inode 349
Connect to /lost+found? yes
Inode 349 ref count is 2, should be 1. Fix? yes
Unattached inode 256714
Connect to /lost+found? yes
Inode 256714 ref count is 2, should be 1. Fix? yes
Pass 5: Checking group summary information
Free blocks count wrong (119473865, counted=119473863).
Fix? yes
Inode bitmap differences: -61
Fix? yes
Free inodes count wrong for group #0 (3668, counted=3669).
Fix? yes
Free inodes count wrong (29485697, counted=29485701).
Fix? yes

rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: 75899/29561600 files (0.2% non-contiguous), 2556223/122030086 blocks

I have found the file of MariaDB is broken before checking it.

dishangti, 2020-10-21:
Though I have make some backups, I still worry about whether the key data files will all lost.

lurch, 2020-10-21:
> the power is supplied by the official power adapter, which can provide 5V 3A, also for the disk

It's always recommended to use either an external power supply or a powered USB hub when using a spinning harddrive, as the Raspberry Pi may not be able to provide enough power over the USB ports. See https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md for more info.

dishangti, 2020-10-31:
> It's always recommended to use either an external power supply or a powered USB hub when using a spinning harddrive, as the Raspberry Pi may not be able to provide enough power over the USB ports. [...]

Thanks for your solution very much. I tried it on my pi, and it has been running stably for a week.

closed 2020-10-31 by dishangti