raspberrypi/Raspberry-Pi-OS-64bit #112: request_module modprobe binfmt-464c cannot be processed

SargonofAssyria, 2020-10-29:
I installed in on a SD-card and it works fine. I have several Raspberry Pi's running from a LUN on my Synology.
I want to do the same with the 64 bit OS. In the SD-card boot environment I can mount the LUN without a problem.
When I try to boot via the SD-card from the LUN is gets into a loop with the above message, which after a view tries fails and then it tries to start /sbin/init but that fails too. Eventually the Raspberry Pi crashes.

pelwell, 2020-10-29:
From https://www.cyber-mage.com/2016/05/06/linux-request_module-runaway-loop-modprobe-binfmt-464c/:
>  Reason:
This usually means you are trying to boot using a 32bit kernel but your installation is 64bit or viceversa.
>
> If you boot a 32-bit kernel with a 64-bit OS, when the kernel tries to start /sbin/init (a 64-bit binary), it won’t recognize the binary format, and it’ll try to load the binfmt-464c kernel module, which is ELF support. (ELF support is generally compiled into the kernel, not built as a module, by the way.)

SargonofAssyria, 2020-10-29:
I did put the official beta image from raspbian on the SD-card. Nothing more, nothing less. So where is the 32-bit kernel with a 64-bit OS of visa versa come from? I only create a initrd.img from the system on the SD-card.
Should I do something extra to include or exclude something from the initrd.img?

SargonofAssyria, 2022-02-11:
In the meantime I have it running. Forgot to close the issue. It is more then 1,5 years ago. Don't know anymore what the solution was.

closed 2022-02-11 by SargonofAssyria