raspberrypi/Raspberry-Pi-OS-64bit #113: Cannot select my screen's native resolution (2560x1440)

warpdesign, 2020-11-03:
I installed the beta from `2020-08-20-raspios-buster-arm64` and when opening the screen configuration utility, I'm stuck to 1920x1080 and cannot select my monitor's resolution which is 2560x1440.

I ran `xrandr` which prints:

> Screen 0: minimum 320 x 200, current 1920 x 1080, maximum: 7680 x 7680
> HDMI-1 connected primary 1920x1080...
> 1920x 1200   59.95
> 1920x1080    60.00*
> 1920x1080i
> 1680x1050
>...

timg236, 2020-11-03:
Please can you check that the HDMI cable is connected to HDMI0 on the Pi4 B

warpdesign, 2020-11-03:
The cable is connected to the HDMI port next to the power port, not sure it's HDMI0.

warpdesign, 2020-11-04:
Just so you know, I had the same problem on the 32bit version a year ago. I opened an issue in the official Pi forums here: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=251248&p=1533537

It was supposed to get fixed in the firmware, but it seems it's still not fixed.

6by9, 2020-11-04:
The use of vc4-fkms-v3d or vc4-kms-v3d means that the thread you reference is irrelevant. That is about how the firmware parses the EDID before the kernel DRM driver takes over.
X in particular ignores any mechanism I know about for trying to influence the mode.

Looking at the EDID that you posted https://www.raspberrypi.org/forums/viewtopic.php?p=1533697#p1533697, the horizontal timings are 2560 active pixels, 47 pixels for front porch, 32 pixels for sync, and 81 pixels for back porch.

As documented at https://www.raspberrypi.org/documentation/configuration/config-txt/pi4-hdmi.md, due to the way that the HDMI pipeline is set up on the Pi4, odd horiztonal timings values can not be supported on the Pi4. This is now enforced by both the firmware and DRM/KMS drivers, therefore it unfortunately will be filtered out of the available modes list.

warpdesign, 2020-11-04:
Does it mean it the native resolution of my monitor does not and will never work correctly on the Pi4?

rtrussell, 2020-11-10:
> odd horizontal timings values can not be supported on the Pi4.

In a case such as this, where both front-porch and back-porch are odd but the sync is even, could is not add one to the front porch and subtract one from the back porch (or vice-versa)?  That would make all the timings even and hopefully do nothing worse than possibly shifting the picture by one pixel.

popcornmix, 2020-11-10:
> could it not add one to the front porch and subtract one from the back porch (or vice-versa)?

That scheme works on many, but not all displays. We used to do that in the firmware driver, but got complaints about displays that weren't happy (i.e. no signal at all). Disabling the resolution so you get a second choice (e.g. 1080p) is a safer option.

lurch, 2020-11-10:
> That scheme works on many, but not all displays.

I wonder if it'd be worth making it a `config.txt` option? :shrug: For those monitors with odd timings where shifting horizontal timings by one pixel _does_ work, adding a boolean `config.txt` option is probably easier than asking the user to enter the complete set of custom timings? Or possibly this is so much of an edge-case that it's not worth bothering with.

6by9, 2020-11-10:
So do you add or subtract? From exactly which timing parameters? There isn't a simple route to fixing up all displays.

@warpdesign You can manually add the mode with a fixup, but there is no way for the the Pi4 to automatically generate a mode which your monitor is guaranteed to be happy with.
Adding
```
hdmi_cvt:0=2560 1440 60
hdmi_group=2
hdmi_mode=87
```
will tell the Pi to use the Coordinated Video Timings (CVT) formula to generate the timings for your display instead of using the EDID. Those typically will work, although the pixel clock tends to be a bit higher than the EDID modes.

rtrussell, 2020-11-10:
> So do you add or subtract? From exactly which timing parameters?

The crucial thing is to keep the total line time unchanged; it probably isn't going to matter much whether you decrement the front porch and increment the back porch or the reverse (one will shift the active picture left one pixel, the other right one pixel).   My background is in broadcast video (both analog and digital) and in that world a monitor being ultra fussy about the front and back porch durations would be considered unacceptable.

warpdesign, 2020-11-10:
> > That scheme works on many, but not all displays.
> 
> I wonder if it'd be worth making it a `config.txt` option? 🤷 For those monitors with odd timings where shifting horizontal timings by one pixel _does_ work, adding a boolean `config.txt` option is probably easier than asking the user to enter the complete set of custom timings? Or possibly this is so much of an edge-case that it's not worth bothering with.

Sounds like a good idea to me.

warpdesign, 2020-11-10:
@6by9 Thanks, I'll try that.

warpdesign, 2020-11-10:
Just wondering: is it a hardware or software problem? Why didn't I encounter any problem with all these devices I plugged into this monitor ?
- MackBook (2017)
- MacBook Pro (2018)
- Mac Mini (2012)
- Surface Book w/dock (2015)

6by9, 2020-11-10:
> The crucial thing is to keep the total line time unchanged; it probably isn't going to matter much whether you decrement the front porch and increment the back porch or the reverse (one will shift [...]

We did a fair amount of investigation on this when it first surfaced. Trust me, there are a fair number of computer monitors that are that fussy.

6by9, 2020-11-10:
> Just wondering: is it a hardware or software problem?

Hardware.

As stated in the [documentation](https://www.raspberrypi.org/documentation/configuration/config-txt/pi4-hdmi.md) I've already linked to, the hardware (pixel valve) generates two pixels per clock cycle, and that includes the timing parameters. That means there is no way to signal that the sync signal changes at an odd pixel boundary.

Originally we had believed that the HDMI block allowed retiming of the sync signals, and therefore allowed regeneration of these odd values, but have been corrected by those at Broadcom who designed the HDMI block.