raspberrypi/Raspberry-Pi-OS-64bit #114: Need Vagrant box for Raspberry Pi OS

nikAizuddin, 2020-11-04:
Hi, can you provide a Vagrant box for Raspberry Pi OS such as [openSUSE Tumbleweed aarch64](https://app.vagrantup.com/opensuse/boxes/Tumbleweed.aarch64)?

pelwell, 2020-11-04:
Why would we want to do that?

nikAizuddin, 2020-11-04:
It would be easy to test development in Vagrant before deploy on Raspberry Pi. For example, I can just change [opensuse/Tumbleweed.aarch64 to Raspberry Pi](https://github.com/extra2000/saferwall-box/blob/3f73f1f32e2e459f58737f33453c087cbaff6815/vagrant/Vagrantfile.saferwall-box.opensuse-tumbleweed.aarch64.example#L24) to test on Raspberry Pi OS.

pelwell, 2020-11-04:
Are you asking us to add support for a virtual Raspberry Pi to Vagrant, something that would mean you don't need a Raspberry Pi to run Raspberry Pi software? I don't think that is going to happen.

nikAizuddin, 2020-11-04:
I see. I thought I could have an easy way to test before deployment with Raspberry Pi OS. Just like other distro.

closed 2020-11-04 by nikAizuddin

lurch, 2020-11-04:
The 64-bit Raspberry Pi OS is based on the arm64 version of Debian, so you should be able to "test" on Debian arm64 before "deploying" on Raspberry Pi OS 64-bit.

nikAizuddin, 2020-11-04:
@lurch Thanks. Unfortunately Debian also doesn't provide official Vagrant box for aarch64. I can't find on Vagrant Cloud. I guess I have to create it then.

lurch, 2020-11-04:
I can never remember how all the different virtualisation / containerisation / emulation platforms fit together, but does https://hub.docker.com/r/arm64v8/debian/ help?