raspberrypi/Raspberry-Pi-OS-64bit #115: Horizontal touch position is wrong when using vc4-fkms-v3d

ereich1217, 2020-11-05:
If the /boot/config.txt contains the following statement it will cause the horizontal position returned from the touch screen to be 2x what it should be:
dtoverlay=vc4-fkms-v3d

Removing the line causes the touch screen to function correctly.  I found this issue with the 7" touch lcd display.

6by9, 2020-11-05:
What version of Pi?
On Pi0-3 composite is enabled if no HDMI display is detected.

In the absence of any other information, X will stretch an input device across the complete desktop, which will include composite.

xinput allows you to configure which display device an input device relates to.
https://www.raspberrypi.org/forums/viewtopic.php?p=1724631#p1724631

lurch, 2020-11-05:
Duplicate of #8 ?

ereich1217, 2020-11-06:
> What version of Pi?
> On Pi0-3 composite is enabled if no HDMI display is detected.
>
> In the absence of any other information, X will stretch an input device across the complete desktop, which will [...]

I am using a Pi3.
It looks very similar. I will have to see if the other behaviors are the same.