raspberrypi/Raspberry-Pi-OS-64bit #117: chromium desktop doesn't start when called with python script.

PIOTA89, 2020-11-07:
Unfortunately the chromium browser does not start if i run the following code: 
- url = 'https://google.at'
- webbrowser.open(url)

If i run the script, nothing happens.

ghollingworth, 2020-11-07:
Just tried it and it worked fine...  There were some error messages but otherwise worked fine

```
pi@raspberry64:~ $ python 
Python 2.7.16 (default, Oct 10 2019, 22:02:15) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import webbrowser
>>> webbrowser.open("https://www.google.com")
True
>>> /usr/bin/xdg-open: 801: /usr/bin/xdg-open: : Permission denied
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to retrieve device information
MESA-LOADER: failed to open kms_swrast (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load driver: kms_swrast
MESA-LOADER: failed to open swrast (search paths /usr/lib/aarch64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri)
failed to load swrast driver

```

PIOTA89, 2020-11-07:
Hi, 
i'am now able to execute this command if i run it without sudo permissions. If you try to run the script with sudp permissions, it doesn't work. Strange....

ghollingworth, 2020-11-07:
And that works fine on 32bit?  It doesn't surprise me that you can't open the browser on a different logged in user's display.

I'm pretty sure that's how it supposed to work

PIOTA89, 2020-11-07:
But for example, I tested this on dietpi 64 Bit beta and it works fine with Firefox. Why not with chromium?

ghollingworth, 2020-11-07:
You should ask the Chromium people.  If it does work on the 32bit Raspberry Pi OS then reopen this issue, but otherwise this is not a 64bit specific problem

closed 2020-11-07 by ghollingworth

PIOTA89, 2020-11-07:
So, i tested this specific problem with raspberry pi 32 bit. And see, there it works! 
But its not a 64 bit issue lel

lurch, 2020-11-08:
For security reasons, it's generally recommended to not run GUI apps as `root`.
But if you _really_ need to, maybe this will point you in the right direction? https://www.google.com/search?q=run+chromium+as+root

PIOTA89, 2020-11-08:
Ok thanks.

raxetul, 2020-11-14:
If the main goal is just running browser from python, there is two option.
1. Runner python app is executed by root(/sudo) and the browser which is run as different user.
2. Runner python app and browser are run as different user except root(/sudo).

Solution for the first one is `os.system("su - <your-user> -c \"<yout-browser>\" ")`
For the the second one, prepare setup to run your app as "`su - <your-user> -c \"<your-app>\"` ".
If yout app is prepared to be a daemon, systemd provides running daemons as user other than root or admins.

I prefer the second one.