raspberrypi/Raspberry-Pi-OS-64bit #119: What's the original cmdline.txt?

misaka19465, 2020-11-15:
I've changed the cmdline.txt, but I can't find the original cmdline.txt. Now I can't boot raspi, can someone give me it?

ghollingworth, 2020-11-15:
console=serial0,115200 console=tty1 root=PARTUUID=dd2cc413-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles

You'll need to replace the PARTUUID with the ID of your drive or "root=/dev/mmcblk0p2"  if you don't know what it is...

closed 2020-11-15 by ghollingworth

lurch, 2020-11-15:
Just for reference, the "upstream source" of `cmdline,txt` is https://github.com/RPi-Distro/pi-gen/blob/arm64/stage1/00-boot-files/files/cmdline.txt (which also gets modified by https://github.com/RPi-Distro/pi-gen/blob/arm64/stage2/01-sys-tweaks/00-patches/07-resize-init.diff and https://github.com/RPi-Distro/pi-gen/blob/arm64/export-image/03-set-partuuid/00-run.sh )

@XECDesign Where does the `plymouth.ignore-serial-consoles` part get added?

XECDesign, 2020-11-16:
That's added by raspi-config when we enable autologin:

https://github.com/RPi-Distro/raspi-config/blob/master/raspi-config#L1554

https://github.com/RPi-Distro/pi-gen/blob/a4be08e275129aea19ba47dd9ff791e1a0037132/stage4/01-console-autologin/00-run.sh#L4