raspberrypi/Raspberry-Pi-OS-64bit #120: Video Playback - Glitches

devshanky, 2020-11-17:
The Video Play back shows lot of glitches. Tried VLC Player and HTML5 player for YouTube. 

File format: MP4. 
Screen resolution: 1920 x 1080. 
Video File Properties: Frame 1280 x 696, Frame rate 24 fps
Device Raspberry Pi 4 / 2 GB RAM. Class 10 SD 16 GB.

Lowering Screen resolution to 1280 x 1024 does not help. Has same issues.

Raspberry Pi-OS-32 bit (armhf) does not have this issue. So probably issue is specific to 64-bit.

popcornmix, 2020-11-17:
Instructions to reproduce this would be useful. Can you play a video from a url using vls or chromium that shows the issue?

ghollingworth, 2020-11-17:
No hardware acceleration, likely just to be a performance problem...

popcornmix, 2020-11-17:
Describing the "glitches" would be help (e.g. provide a picture).
Do some videos work (e.g. lower resolution or different format)

devshanky, 2020-11-18:
The "glitches" were more like electrical signal interference on the screen, but very brief and intermittent. It was definitely not like frame drops but persistent.

The same media file works flawless on RPi OS 32-bit on both RPi 3+ and RPi 4 regardless of screen resolution. I also checked SD Card and it passes all tests on RPi 4.

Suddenly I have lost HDMI output on my RPi 4. I tried EEPROM recovery but it doesn't work so I am getting a new micro-HDMI cable. I will be able to post pictures once I get the hardware.

lurch, 2020-11-18:
"The "glitches" were more like electrical signal interference on the screen" ... "I am getting a new micro-HDMI cable"

I wonder if these things might be related? :wink:

devshanky, 2020-11-19:
I got a brand new cable and the HDMI is working now! 

I repeated the tests and found same results as before. The glitches are quite visible on RPi OS 64 bit but does not happen on 32 bit.

Here is a link to the video capture from 64 bit version. The sound is muted for the demo.

https://drive.google.com/file/d/1fposbm6k-OaIajgE1vdksfs79zLArfEI/view?usp=sharing

[](https://drive.google.com/file/d/1fposbm6k-OaIajgE1vdksfs79zLArfEI/view?usp=sharing)

devshanky, 2020-11-19:
I played few other movies with same results. I also tried playing this using MPV and still see the glitches. Used VLC for earlier tests.
LiberELEC / Kodi plays these movies just fine on RPi 4.
It is quite possible this is a software issue, probably drivers.

lurch, 2020-11-19:
Ahhh, by "electrical interference" I thought you meant speckles on the display or something, but i guess you're actually referring to the tearing, like in the top half of this screenshot from your video?

[image]

devshanky, 2020-11-19:
More like it. I apologize it is hard to describe this for me. They appear randomly every few moments all over the screen at random locations. Unable to pin a specific pattern to this.

popcornmix, 2020-11-19:
Tearing in desktop is a known issue. Running fullscreen should avoid it.

devshanky, 2020-11-19:
Full screen @ 1920 x 1080 does not help. However Full screen @ 1280 x 1024 @ 60 Hz improves the output a lot but I still see this occasionally. Just wondering if this is a driver issue.

popcornmix, 2020-11-19:
How are you switching to fullscreen? 'f' key in vlc should do it.

devshanky, 2020-11-19:
I tried both, did a right click and picked the menu option "Fullscreen" and use F11 to go to Fullscreen.

devshanky, 2020-11-23:
**Issue resolved:**
Started with image **2020-08-20-raspios-buster-arm64-lite.img** and installed just the basics:
**xserver-xorg-core xserver-xorg xfonts-base xinit xfce4 desktop-base lightdm xfce4-terminal vlc chromium**

After this tearing has almost vanished. It now works like charm. Looks like the default lxde desktop might have some components that is causing some interference.

closed 2020-11-23 by devshanky