raspberrypi/Raspberry-Pi-OS-64bit #122: Pyxel library install fail

bifft2, 2020-12-09:
The backstory is that I had trouble installing pyxel on the 32 bit OS, so I decided to try on the 64 bit OS. 

When I install the pyxel library with `sudo -H pip3 install -U pyxel` and then try to import pyxel, I get the following error:

Traceback (most recent call last):
  File "/home/pi/Documents/pyxelTest.py", line 1, in <module>
    import pyxel
  File "/usr/local/lib/python3.7/dist-packages/pyxel/__init__.py", line 10, in <module>
    from . import core  # type: ignore
  File "/usr/local/lib/python3.7/dist-packages/pyxel/core/__init__.py", line 42, in <module>
    _lib = _load_library()
  File "/usr/local/lib/python3.7/dist-packages/pyxel/core/__init__.py", line 39, in _load_library
    return cdll.LoadLibrary(lib_path)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: /usr/local/lib/python3.7/dist-packages/pyxel/core/bin/linux/libpyxelcore.so: cannot open shared object file: No such file or directory

From what I can tell, that file exists in that location. Any help would be much appreciated!

lurch, 2020-12-09:
If it doesn't work on either the 32-bit OS or the 64-bit OS (and it's also also only related to 3rd-party software rather than the OS itself), then it sounds like this is a question better suited for the [forums](https://www.raspberrypi.org/forums/) than this issue-tracker?

lurch, 2020-12-09:
https://github.com/kitao/pyxel#other-environment ?

closed 2020-12-09 by pelwell

bifft2, 2020-12-09:
Yeah, I tried the Other Environment install, but also had trouble there. See [this issue](https://github.com/kitao/pyxel/issues/286). Would love some ideas for where to go next. Thank you!

lurch, 2020-12-09:
Filing an issue on the bug-tracker for the software you're having problems with, sounds like **exactly** the right thing to do. Hopefully they'll be able to help.