raspberrypi/Raspberry-Pi-OS-64bit #123: libcrypt.so.1: version `XCRYPT_2.0' not found

tdiesler, 2020-12-09:
This happens when I try to run [monit](https://mmonit.com/monit)

```
pi@raspi4:~$ monit -t
monit: /lib/aarch64-linux-gnu/libcrypt.so.1: version `XCRYPT_2.0' not found (required by monit)
```

Also and perhaps related ...

```
pi@raspi4:~$ sudo apt install monit
...
Package monit is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
E: Package 'monit' has no installation candidate
```

ghollingworth, 2020-12-09:
Monit isn't in the ARM64 Debian build, you'll have to install it from sources

closed 2020-12-09 by ghollingworth

lurch, 2020-12-09:
Looks like it's in buster-backports, but not in buster itself https://packages.debian.org/search?keywords=monit

tdiesler, 2020-12-09:
This issue may have been closed prematurely. I'm not necessarily complaining about monit not being available in the apt repo. I'm saying, when you install monit manually and then run ...

```
monit -h
```

you get 

```
/lib/aarch64-linux-gnu/libcrypt.so.1: version `XCRYPT_2.0' not found
```

The monit arm64 distribution is [here](https://www.mmonit.com/monit/#download)

XECDesign, 2020-12-09:
Works fine if you built it from source. The issue seems to be with their binary builds.

lurch, 2020-12-09:
Yeah, when you download a pre-compiled Linux binary, you have no way of knowing what libraries it was linked against when it was built; and if you're unlucky (as you are here) it might have been linked against different library-versions than you have installed on your local system.

tdiesler, 2020-12-09:
Yes, works great