raspberrypi/Raspberry-Pi-OS-64bit #124: No memory accounting with `docker stats`

tdiesler, 2020-12-11:
```
# docker stats

CONTAINER ID   NAME      CPU %     MEM USAGE / LIMIT   MEM %     NET I/O          BLOCK I/O        PIDS
7ec7cbd0c3a2   test      129.45%   0B / 0B             0.00%     23.5MB / 4.4MB   61.4kB / 917MB   136
```

Docker info reports

```
Client:
 Context:    default
 Debug Mode: false
 Plugins:
  app: Docker App (Docker Inc., v0.9.1-beta3)
  buildx: Build with BuildKit (Docker Inc., v0.4.2-docker)
Server:
 Containers: 1
  Running: 1
  Paused: 0
  Stopped: 0
 Images: 14
 Server Version: 20.10.0
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Native Overlay Diff: true
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 io.containerd.runtime.v1.linux runc
 Default Runtime: runc
 Init Binary: docker-init
 containerd version: 269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc version: ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 init version: de40ad0
 Security Options:
  seccomp
   Profile: default
 Kernel Version: 5.4.79-v8+
 Operating System: Debian GNU/Linux 10 (buster)
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 7.633GiB
 Name: raspberrypi
 ID: TE3N:EJS2:5RYW:GCIP:GWD3:YORB:65HE:FMC6:YEXV:YOIW:OWLE:LKVL
 Docker Root Dir: /var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Labels:
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false
WARNING: No memory limit support
WARNING: No swap limit support
WARNING: No kernel memory TCP limit support
WARNING: No oom kill disable support
WARNING: No blkio weight support
WARNING: No blkio weight_device support
```

Similar issues have been found with Debian before and it seems to have to do with missing memory accounting in cgroup

```
pi@raspberrypi:~ $ sudo ls -l /sys/fs/cgroup
total 0
dr-xr-xr-x 6 root root  0 Feb 14  2019 blkio
lrwxrwxrwx 1 root root 11 Feb 14  2019 cpu -> cpu,cpuacct
dr-xr-xr-x 6 root root  0 Feb 14  2019 cpu,cpuacct
lrwxrwxrwx 1 root root 11 Feb 14  2019 cpuacct -> cpu,cpuacct
dr-xr-xr-x 3 root root  0 Feb 14  2019 cpuset
dr-xr-xr-x 6 root root  0 Feb 14  2019 devices
dr-xr-xr-x 3 root root  0 Feb 14  2019 freezer
lrwxrwxrwx 1 root root 16 Feb 14  2019 net_cls -> net_cls,net_prio
dr-xr-xr-x 3 root root  0 Feb 14  2019 net_cls,net_prio
lrwxrwxrwx 1 root root 16 Feb 14  2019 net_prio -> net_cls,net_prio
dr-xr-xr-x 3 root root  0 Feb 14  2019 perf_event
dr-xr-xr-x 6 root root  0 Feb 14  2019 pids
dr-xr-xr-x 6 root root  0 Feb 14  2019 systemd
dr-xr-xr-x 5 root root  0 Feb 14  2019 unified
```

pelwell, 2020-12-11:
The memory cgroup isn't enabled by default - make sure that `cgroup_memory=1 cgroup_enable=memory` is in cmdline.txt (as part of the single line).

tdiesler, 2020-12-11:
Yes, that works - thanks.

closed 2020-12-11 by tdiesler