raspberrypi/Raspberry-Pi-OS-64bit #125: Cannot change cpu frequency

kpg02, 2021-01-09:
I'm trying to OC my rpi 4b and I can't the max frequency it achieves is 1.5 GHz, it's the max set by the system, I'm trying to put it to 2GHz, but my changes on /boot/config.txt don't make effect, I have changed the #arm_frewq=700 to
 
over_voltage=6
arm_freq=2000
gpu_freq=750

without any effect (Of course, I did reboot)

popcornmix, 2021-01-10:
Can you report output of
```
vcgencmd version
uname -a
vcgencmd get_config int
```

kpg02, 2021-01-10:
@popcornmix  

Jan  7 2021 18:27:29

Copyright (c) 2012 Broadcom

version fb345a0c2d5544957f4ba1a2b9e968970e3312c4 (clean) (release) (start)

--
Linux raspberrypi 5.10.5-v8+ #1391 SMP PREEMPT Thu Jan 7 17:55:54 GMT 2021 aarch64 GNU/Linux

--
arm_64bit=1
arm_freq=2000
audio_pwm_mode=514
config_hdmi_boost=5
disable_commandline_tags=2
disable_l2cache=1
display_hdmi_rotate=-1
display_lcd_rotate=-1
enable_gic=1
enable_uart=1
force_eeprom_read=1
force_pwm_open=1
framebuffer_ignore_alpha=1
framebuffer_swap=1
gpu_freq=750
gpu_freq_min=250
init_uart_clock=0x2dc6c00
lcd_framerate=60
mask_gpu_interrupt0=1024
mask_gpu_interrupt1=0x10000
max_framebuffers=2
over_voltage=6
over_voltage_avs=-10000
pause_burst_frames=1
program_serial_random=1
total_mem=8192
hdmi_force_cec_address:0=65535
hdmi_force_cec_address:1=65535
hdmi_pixel_freq_limit:0=0x11e1a300
hdmi_pixel_freq_limit:1=0x11e1a300

popcornmix, 2021-01-11:
Seems to b e working for me. I have:
```
arm_64bit=1
arm_freq=2000
gpu_freq=750
over_voltage=6
```
and:
```
pi@pi4:~ $ echo performance | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
performance
pi@pi4:~ $ vcgencmd measure_clock arm
frequency(48)=2000478464
pi@pi4:~ $ echo powersave | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
powersave
pi@pi4:~ $ vcgencmd measure_clock arm
frequency(48)=600117184
pi@pi4:~ $ echo ondemand | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
ondemand
pi@pi4:~ $ vcgencmd measure_clock arm
frequency(48)=1400308608
```
When using ondemand governor the arm frequency varies between 600MHz and 2000MHz according to load.
Can you explain exactly how this is not working for you?

kpg02, 2021-01-11:
@popcornmix 

```
pi@raspberrypi:~ $ echo performance | sudo tee /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
performance
pi@raspberrypi:~ $ vcgencmd measure_clock arm
frequency(48)=1500345728
```

pelwell, 2021-01-11:
To rule something out, how about `vcgencmd get_throttled`?

kpg02, 2021-01-11:
@pelwell 
throttled=0x0

popcornmix, 2021-01-11:
Were you getting 600 or 1500 with powersave governor?
What about:
```
pi@pi4:~ $ cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_available_frequencies 
600000 700000 800000 900000 1000000 1100000 1200000 1300000 1400000 1500000 1600000 1700000 1800000 1900000 2000000 
```

kpg02, 2021-01-11:
@popcornmix 

1500

same output as you

popcornmix, 2021-01-12:
If you have a spare sdcard it might be worth checking if you have the same issue on a fresh install of RPiOS.
That would identify if some setting on the sdcard is restricting it (but I can't immediately think what).