raspberrypi/Raspberry-Pi-OS-64bit #126: Wireless Network support break after connecting USB Adruino Uno

gartlana, 2021-01-11:
Problem statement: 
'No Network interface available' with any Raspberry OS 64 Beta Micro-SD build. New and different date beta build files all say Network interface unavailable across numerous vanilla reboots and new configuration wizard workflows.

Problem scenario:

Step1: Target build created and all initially fully functional including wireless networking as follows: 
Software:
  O/S Source: 2020-08-20-raspios-buster-arm64.zip source build on 64Gb Micro-SD
  Application: Universal GCode Sender

Result(s): Network including internet access all working o.k. 
                 Application interface stable and functionality partially verified o.k.

Step2: Connect Adruino Uno via USB
  Hardware: Pi400
      Adruino Uno connected via USB (proven  via Rasperry OS 32bit build prior and following this exercise)

Results(s): Universal GCode Sender did not display serial device for Adruino.
                   O/S Network interface remained available and functioning to provideing internet access.

Step 3: Reboot with Adruino connected via USB

Results(s): No network interface available.
Subsequent reboots and also attempts with different Raspberry OS 64 Beta Micro-SD  source/date beta vanilla builds all say Network interface unavailable through the new configuration wizard workflow. Interestingly rebooting at any point with 32bit current Buster version sees Universal GCode Sender with Adruino UNO via USB and network interface all fully functional concurrently, but switching to any new 64bit build fails to detect a network interface (i.e. the problem appears to 'stick' to 64bit OS builds).

FYI - Final aim is a platform build with Raspian/Raspberry OS, Universal GCode Sender and FreeCad functioning for a 3-axis CNC. 

I am new to Pi/Linux/Debian/etc so please excuse if this is not helpful. 
Respectfully, Andrew

pelwell, 2021-01-11:
The output from `dmesg` with the Arduino attached would help in the diagnosis, but if I had to guess I would say it's power related. Booting often has the largest power drain, with many subsystems all being started in parallel. I've seen many instances where one-too-many USB devices prevents the WiFi from starting up, even though just about the only things they share are the processor cores and the power supply.

The Pi 400 needs a solid 3A @ 5.1V. Note that phone chargers often don't make good power supplies because it doesn't matter if their voltage drops from time to time. USB devices that are anything more than keyboards, mice and SD card readers should probably have independent power supplies or be on a powered hub.

gartlana, 2021-01-11:
Hi Phil,
Thank you I will try to capture dmesg following a reboot tomorrow ( I need to research the method(s)).
FYI’s - A standard Raspberry Pi supplied power supply is in use.
          - I do not need the Adruino connected during boot to recreate the problem (I.e whatever triggered the problem now persists across boots and power on/off scenarios with or without the Adruino UNO connected).
          -  The Adruino was USB powered in these scenarios as you suggest. I will source a separate power supply before reconnecting it.
          - Connecting the Adruino UNO after booting, powered by USB, in the 32bit Raspberry OS builds did not trigger the problem, but I will avoid booting with USB powered Adruino in future. I don’t recall if I ever booted with the Adruino USB connected in the 32bit builds.
          - I am wondering if there is something in the pi400 (e.g firmware/bios) that might have been corrupted and can be reinitialised somehow but currently lack the knowledge to explore and validate that theory).

Andrew

pelwell, 2021-01-11:
I don't want to speculate too much without seeing the kernel log, but the chances of this behaviour being due to corruption seem extremely small; SD card corruption is likely to manifest in more unpleasant ways, and EEPROM corruption is virtually impossible and likely to be catastrophic. At the moment it does seem more likely that some apparently insignificant software change has caused this problem.

gartlana, 2021-01-11:
Hi Phil,
Thanks. Appreciate the methodical approach and additional information.
The dmesg out text file I have now (hopefully) attached contains three boot scerio dmesg outputs: 
1. Boot of vanilla 64bit Raspberry OS on Pi400 without USB/Adruino_UNO
2. Boot of same vanilla 64bit Raspberry OS on Pi400 with USB/Adruino_UNO attached during boot
3. Boot of current 32bit Raspberry OS Buster on Pi400 without USB/Adruino_UNO, but able to attach and use UNO with Universal GCODE Sender in parallel with wireless network interface concurrently with all o.k.

[dmesg_2021_Jan_12_NoWirelessInterface_Available_1.txt](https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/files/5797978/dmesg_2021_Jan_12_NoWirelessInterface_Available_1.txt)

Respectfully, Andrew

gartlana, 2021-01-11:
Additional information only:
Same pi400 when booted from Micro-SD containing current Ubuntu Desktop 64bit OS has working wireless network connectivity and appears to recognise the UNO when connected via USB after boot as per the following diagnostics:
 
andrew@andrew-desktop:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04d9:0007 Holtek Semiconductor, Inc. 
Bus 001 Device 003: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 005: ID 2341:0043 Arduino SA Uno R3 (CDC ACM)
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
andrew@andrew-desktop:~$ 

The application ugsplatform "Universal GCode Sender" doesn't recognise the UNO on this platform build and while I assume that is outside scope and potentially very different issue(s), I saw some reference to Ubuntu in Raspberry Pi dmesg output so thought it might be interesting/helpful to see how current 64bit Ubuntu behaves in the same scenario.

pelwell, 2021-01-11:
```
[   32.751255] brcmfmac: brcmf_sdio_htclk: HT Avail timeout (1000000): clkctl 0x50
```
This line indicates an unresponsive WiFi controller, which can be caused by something interfering with the GPIO expander and pulling it's enable line low, but is more likely to be caused by undervoltage. Chips built on the same production line vary in the way they consume power, and this can lead to problems in different parts of the chip that can't be detected by the crude voltage detection circuit.

Try adding a small overvoltage by adding `over_voltage=2` to config.txt.

gartlana, 2021-01-11:
Thanks. Updates as follows:
Step1: Added over-voltage=2 and reboot.
Result(a): No change / No Wi-Fi 

Step2: Added over-voltage=6 and reboot.
Result(a): No change / No Wi-Fi 

I’m going to try to recreate a current vanilla 64bit latest Beta build Micro-SD on the iMac exactly as I did the original image which worked, with wireless, up until I attempted to connect the ugsplatform application to the UNO and will check back with you.

Question: How can verify that the over-voltage change took effect?

Andrew Gartlan. 
Andrew.gartlan@me.com

gartlana, 2021-01-11:
After:
1. the Raspberry Pi OS latest 64bit Micro-SD was reformatted fat32
2.  Then boot image recreated using the Raspberry imager as per doc.
3. Reboot Pi400 from MicroSD with no USB devices attached.

Result(s):
1. No Wi-Fi. Hovering over the ‘broken’ Wi-Fi widget icon symbol display msg “eth0: Link is down”.

On 12 Jan 2021, at 7:47 am, Andrew Gartlan <andrew.gartlan@me.com> wrote:

Thanks. Updates as follows:
Step1: Added over-voltage=2 and reboot.
Result(a): No change / No Wi-Fi 

Step2: Added over-voltage=6 and reboot.
Result(a): No change / No Wi-Fi 

I’m going to try to recreate a current vanilla 64bit latest Beta build Micro-SD on the iMac exactly as I did the original image which worked, with wireless, up until I attempted to connect the ugsplatform application to the UNO and will check back with you.

Question: How can verify that the over-voltage change took effect?

Andrew Gartlan. 
Andrew.gartlan@me.com

gartlana, 2021-01-11:
Hi Phil,
I'm wondering about corruption of the firmware on the pi400 still. Ref. https://jamesachambers.com/raspberry-pi-4-bootloader-firmware-updating-recovery-guide/ 
Your thoughts appreciated :)

gartlana, 2021-01-11:
FYI - Current Pi400 EEPROM firmware level follows (captured using 32bit Buster environment):
pi@raspberrypi:~ $ sudo rpi-eeprom-update
[sudo] password for pi: 
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/critical
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1

JamesH65, 2021-01-12:
> Thanks. Updates as follows: Step1: Added over-voltage=2 and reboot. 

That should be over_voltage, not over-voltage.  (underscore vs hyphen)

gartlana, 2021-01-12:
Hi Phil,
I confirm that I tested with "over_voltage=" values, however I incorrectly typed the verb in my update above (my apology).

pelwell, 2021-01-12:
It's not obvious what might be wrong. One option would be for you to upgrade the 64-bit kernel to the current builds, which are based on Linux 5.10 (the next Long Term Stable kernel). We don't recommend the use of pre-release kernels and firmware without good reason, but I think this constitutes a good reason.

Upgrade the firmware and kernel using `sudo rpi-update`. If you can make a backup of the SD card before you update then I would recommend it, just in case something goes wrong.

gartlana, 2021-01-12:
Hi Phil,
Test_Firmware_1:
1. Establish Internet connectivity: Connected Pi400 with 64bit Beta Raspberry OS to the internet via an Ethernet cable (as wireless is not working on the 64bit Beta OS build).
2. Ran sudo rpi-update.
Results(s): Connected and ran error free.
3. Disconnect Ethernet connection and reboot (no USB devices attached).
Result(s): No change / Wireless connectivity unavailable.

FYI's:
Additional verification performed:
a) I wanted to know if the 32bit OS had any issues with the UNO USB connected during a boot to help isolate the issue towards hardware/firmware/software so prior to the above "rpi update" I cloned my 32bit Raspian Buster SD (with over_voltage=6) and rebooted with Adruino Uno connected via USB in an attempt to recreate the trigger event which appeared to correlate with the loss of wireless connectivity in the 64Bit Beta build (remembering that initially the 64bit Beta image wireless was working).
Results: The 32bit image has no problem with the UNO connected and powered via USB  on this Pi400 hardware. I am not confident to draw any firm conclusions from this test but hope share in case it helps(s) others.

Please note: I have two identical 32bit Raspian buster SD's that work fine on this Pi400 and am happy to try things on the Beta 64bit SD(64Gig). At this point I'm still not sure if there is a problem 'sticking' to the 64bit build and my specific Pi400 hardware, or simply a bug in the 64bit software which I would expect others to find and report also.

Hope this proves helpful, Andrew

gartlana, 2021-01-12:
Test_APT_UPDATE1:
With the Pi400 64bit image now internet connected via Ethernet I can now run the usual SUDO APT UPDATE (207 packages can be upgraded) and sudo apt full-upgrade.
Result(s): I will update if there is any change/improvement following reboot.

gartlana, 2021-01-13:
Hi Phil,
Status Update: Service (wireless support) restored.
Method: 
1. Created a new 64bit Beta Micro_SD (to reset the baseline only).
2. Connected the Pi400 to the internet via Ethernet
3. Rebooted from the newly created 64bit Beta Micro-SD
Result(s):
a. Lots of updates were able download during the 'Welcome to Pi' configuration process
b. On completion of the 'Welcome to Pi' startup configuration process, I was able to access "Wireless & Wired Network settings" which triggered a prompt to specify country/region (n.b this had already been specified earlier in the Welcome script but I needed to do it again specifically for this first-time wireless setup). Upon specifying Country=Australia I was able to enable wireless and attach  to my home access point. Access to wireless settings/configuration/service restored :) 

Problem status: I will try to recreate/confirm the problem trigger (either booting with the UNO USB attached while using wireless concurrently and/or running Univeral GCode Sender ugsplatform in this configuration and advise if/when I find something more.

Comments: As far as recovering from a corrupted image is concerned, the 'Welcome to Pi' script seems to need internet connectivity to complete normally, else unpredictable results may result. I believe this is clearly stated in the manual so "Read the Manual" is probably my only valuable lesson for 64bit Beta testing problem recovery paradigms at this point ... 

Thanks for the assist Phil. Much appreciated.

gartlana, 2021-01-13:
FYI - Have now also successfully rebooted the vanilla 64bit Beta image with wireless connectivity and USB attached and powered Adruino UNO (so no over_voltage= override specified) throughout. No problem was triggered with the now configured wireless connectivity. The only untested element in the original problem scenario that I'm aware of is the use of the ugsplatform application which I assume belongs in another forum.

Phil please close this if/when you are happy there is no more value here for the Beta program.

closed 2021-01-13 by pelwell