raspberrypi/Raspberry-Pi-OS-64bit #128: File manager and desktop preferences are not being saved

jharris1993, 2021-01-16:
Greetings!

Experimenting with the August release of Raspberry-Pi-OS-64bit, I noticed that certain GUI preferences are not being saved.

In particular, desktop preferences do not persist past a reboot, and the ability to change the wallpaper doesn't work at all - you cannot change it at all.

Likewise the file manager preferences - particularly the preference to "mount mountable volumes at startup" does not persist.  If I clear it, it stays cleared until I reboot then it comes back selected and all mountable volumes automount.

I have not investigated other preferences, but I would not be surprised if other GUI preferences are not being saved as well.

The recent versions of the 32 bit Raspberry Pi O/S do not exhibit these issues.

XECDesign, 2021-01-18:
Is anything being saved? Is it possible the storage medium is the problem. If you create a new file and reboot, is it still there?

Because I'm running a 64-bit image with a changed wallpaper, moved taskbar and so on without any issues.

lurch, 2021-01-18:
@jharris1993 Have you accidentally enabled the read-only rootfs option in raspi-config?

MarkDH102, 2021-01-27:
I use Claws mail and the certificates are not being saved. Every time I launch claws and/or reboot I keep getting the prompt for "new certificates".
Also, the weather widget on the taskbar seems unable to start up from boot and find the weather, it shows a yellow exclamation mark. I have to go into the widget settings (which still has my correct location) and click OK at which point the task bar updates with the current weather. I don't know if this may also be related to a not saving settings scenario...
Other files that I create in my user space ARE being saved correctly. I have set desktop preferences that have been correctly saved too. So it isn't my SD card...