raspberrypi/Raspberry-Pi-OS-64bit #129: Cannot compile https://github.com/cilynx/rtl88x2bu

xMadNet, 2021-01-18:
Linux XPiRasp4 5.10.7-v8+ #1394 SMP PREEMPT Thu Jan 14 20:32:25 GMT 2021 aarch64 GNU/Linux
Rpi 4b 4Go
raspberrypi-kernel-headers is already the newest version (1.20201201-1).

Compile is Ok
BUT
ERROR: modpost: "__const_udelay" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_from_user" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "_raw_spin_unlock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_lock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__rcu_read_unlock" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "cpu_hwcap_keys" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "csum_ipv6_magic" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__warn_printk" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "arm64_const_caps_ready" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__udelay" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "preempt_schedule_notrace" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
ERROR: modpost: "__arch_copy_to_user" [/home/pi/wifidrivers/rtl88x2bu/88x2bu.ko] undefined!
m###

lurch, 2021-06-15:
If code from an external repo doesn't work on the Raspberry Pi OS 64-bit beta, it's probably the job of that external repo to fix their code to be compatible? :man_shrugging:

guille-mp, 2021-06-15:
Sorry, i copied the error from another repository but the error is the same.