raspberrypi/Raspberry-Pi-OS-64bit #130: Ethernet doesn't work after updating

garry200708, 2021-02-01:
On RPi 4B 4GB
Steps to recreate this problem:
1 install and boot
2 open terminal and do apt update & upgrade
3 reboot

ghollingworth, 2021-02-01:
Can't reproduce, what does ipconfig say?

garry200708, 2021-02-01:
I've just deleted RPiOS from SD card, but eth0 is missing in ifconfig.

garry200708, 2021-02-01:
WiFi works perfectly

blacklat, 2021-04-17:
I bought my rpi4 in 2019 (2 years ago) and there is still no official 64-bit stable OS, this is extremely shameful for the raspberry folks and for customers it is extremely frustrating and annoying. I have no other words to describe the raspberry coders than "incompetent". The folks at retro game systems are years ahead of the raspberry folks. This only shows that the raspberry team is composed only of people without technical knowledge, amateurs, cheap labor.
One thing has to be said and understood by them, this shit is not free, I paid for it and paid dearly in the pre-launch, we are consumers who pay their wages, the operating system is their obligation to be made available for free to everyone who purchased an rpi board. If your employees lack the intelligence to build a stable system, then send them away and hire qualified people. I bought a 64-bit rpi board, fuck the 32-bit OS, make the fucking 64-bit OS work, be it professional, lamers.