raspberrypi/Raspberry-Pi-OS-64bit #131: xdg-autostart not present on image

chris-y, 2021-02-06:
raspi-config has an option to enable xcompmgr for compositing.  This adds an entry in /etc/xdg/autostart/, which is a directory xdg-autostart uses to run programs at login (there are also other files in there).
However, xdg-autostart is not installed by default!

The package "obsession" needs adding to the image, and then xdg-autostart needs to be added to the default autostart(?).
I tried installing and running xdg-autostart manually but it doesn't appear to do much other than spew out some errors, so I think the files already in /etc/xdg/autostart also need checking.

lurch, 2021-02-07:
ping @XECDesign & @spl237

spl237, 2021-02-07:
I've just tried the latest arm64 image, and autostart is working correctly. If it wasn't, the first-run wizard wouldn't be displayed - it is launched by the piwiz.desktop file in /etc/xdg/autostart - and it appears on first boot as it should.

There are numerous parts of the desktop which require autostart - if it wasn't working, most of the desktop wouldn't appear! You shouldn't need to install anything to make it work - it works out of the box, and xdg-autostart isn't needed.

chris-y, 2021-02-07:
I've just done some further testing and, yes, it does work.  What threw me is that xcompmgr isn't running after login.  I've now tracked this down to the ifs in the script it runs to launch:
```
if grep -q okay /proc/device-tree/soc/v3d@7ec00000/status 2> /dev/null || grep -q okay /proc/device-tree/soc/firmwarekms@7e600000/status 2> /dev/null ; then
    xcompmgr -aR
fi
```

The first of those files doesn't exist, the second one reports "disabled".  I'm using full KMS, so maybe there are additional checks it needs to do here?

lurch, 2021-02-07:
ping @pelwell for further info about DeviceTree status files :slightly_smiling_face:

XECDesign, 2021-02-08:
There's also the question of whether we should be using xcompmgr with full KMS (or at all). In my experience, it just reduces performance. IIRC, the original reason it was added was that it reduced tearing when dragging windows?

pelwell, 2021-02-08:
The fragment of shell script above should be true in the following situations:
1. With `dtoverlay=vc4-fkms-v3d` on any Pi.
2. With `dtoverlay=vc4-kms-v3d` on a Pi with BCM2835, BCM2836 or BCM2837, i.e. not BCM2711.