raspberrypi/Raspberry-Pi-OS-64bit #132: eth0 randomly "state DOWN" on boot

bunjg, 2021-02-09:
My Pi is set up with a script that runs on boot which reconfigures the speed of eth0 to be set to 1000 (via “ethtool -s eth0 speed 1000 Mb/s”). Occasionally I’ve noticed on some boots (not every time) that eth0 will not come up when configured with speed 1000 Mb/s. 

When this occurs, ethtool reports “Link detected: no”, and “ip a” returns “<NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000”. The lights on the RJ45 jack also do not come on when this issue occurs.

Cycling the adapter down/up via “sudo ip link set eth0 down” & “sudo ip link set eth0 up” does not correct the issue.

Strangely, changing the speed of eth0 to 100 Mb/s (in lieu of 1000) seems to cause the adapter to work again (lights come on, ethtool reports link detected, and “ip a” reports that state changes to UP), but reverting back to 1000 Mb/s causes the link to go down again. The only thing that seems to fix this is a reboot, though I’d say that the issue seems to recur every 3 – 5 boots approximately.

I’ve tested this on with a different SD card (both running the latest 64-bit Pi OS), and also tested a different Pi, and had the same results with both. I also used to run Ubuntu 64-bit on my Pi’s before the 64-bit Pi OS was available, and never had the issue with that OS, which leads me to believe the issue is within the 64-bit Pi OS.

lurch, 2021-02-09:
We had similar reports in #57 , #91 and #130
Do any of those contain information which help you diagnose or resolve your problem?
(and can you please try testing with the 32-bit version of Raspberry Pi OS too, to verify whether it's a 64-bit only issue or a more general Raspberry Pi OS issue?)

And just out of curiosity, why are you manually setting a 1000 Mb/s speed? Shouldn't it autonegotiate?

bunjg, 2021-02-11:
Tried #57 and it didn't help.  #91 doesn't match because that was an erratic failure, where in my case , if you try to set eth0 to speed=1000, duplex=full, autoneg=off, AND you get a state of DOWN, it never recovers until another reboot.   #130 doesn't seem to match because eth0 is always visible.

Will try to reproduce the issue on 32-bit version OS, but I need the 64-bit for the final product (that's why I was previously using Ubuntu).  I had some issues with autonegotiate earlier in the project, which setting the speed resolved.  I am using some different hardware now, so I will see if the old issue, or new this current issue, appear if I drop the speed configuration from my script.

BitBistro-code, 2021-04-06:
I am running a pi4 8g 64bit and gige with no issues. It is highly unlikely that such a basic function like the built in nic was not tested internally. It is much more likely that this is an issue with your cable and/or your switch.
You should not need to specify a speed and duplex on gige. Autoneg is intrinsic to the spec. Try not hard coding it, replace your cable, try another port or your switch, and look for errors or misconfiguration on the switch side if it is managed.

blacklat, 2021-04-17:
I bought my rpi4 in 2019 (2 years ago) and there is still no official 64-bit stable OS, this is extremely shameful for the raspberry folks and for customers it is extremely frustrating and annoying. I have no other words to describe the raspberry coders than "incompetent". The folks at retro game systems are years ahead of the raspberry folks. This only shows that the raspberry team is composed only of people without technical knowledge, amateurs, cheap labor.
One thing has to be said and understood by them, this shit is not free, I paid for it and paid dearly in the pre-launch, we are consumers who pay their wages, the operating system is their obligation to be made available for free to everyone who purchased an rpi board. If your employees lack the intelligence to build a stable system, then send them away and hire qualified people. I bought a 64-bit rpi board, fuck the 32-bit OS, make the fucking 64-bit OS work, be it professional, lamers.