raspberrypi/Raspberry-Pi-OS-64bit #135: rpi case fan pios 64bit (gpio-fan?) does not start the fan

cyclic99, 2021-02-20:
rpi case fan pios 32bit with desktop works i.e fan starts at trigger temp ok.
pios 64bit LITE does not work;  goes through trigger temp onwards until throttling.
raspi-config seems to do it's setup ok; I have tried various temps & clocks.
I am occasional seeing a twitch on the fan passing through trigger temp.
Once it ran for one second then stalled.

pelwell, 2021-02-20:
Are you definitely running the same kernel version (`uname -a`) and firmware version (`vcgencmd version`) in both cases?

cyclic99, 2021-02-21:
Desktop 
Linux tb 5.10.11-v7l+ #1399 SMP Thu Jan 28 12:09:48 GMT 2021 armv7l GNU/Linux
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)
LITE 
Linux z1 5.10.11-v8+ #1399 SMP PREEMPT Thu Jan 28 12:14:03 GMT 2021 aarch64 GNU/Linux
version 99d9a48302e4553cff3688692bb7e9ac760a03fa (clean) (release) (start)

pelwell, 2021-02-21:
Thanks - I'll give it a try.

cyclic99, 2021-02-21:
Incidently I notice  lshw  show deltas :
Desktop @32bit  -
    description: ARMv7 Processor rev 3 (v7l)
    product: Raspberry Pi 4 Model B Rev 1.2
    serial: 10000000965674b7
    width: 32 bits
LITE @64bit  - 
    description: Computer
    product: Raspberry Pi 4 Model B Rev 1.1
    serial: 1000000079294cf2
    width: 64 bits
Both are 4GB
i.e. Might Rev be "relevant"?

pelwell, 2021-02-21:
Board revision should not be relevant. Was this an intentional close?

cyclic99, 2021-02-21:
Reopened - closed in error - mouse lag..

pelwell, 2021-02-22:
Firmware release 99d9a483 was made over a week before the PoE HAT fix went in - running `sudo SKIP_KERNEL=1` should make it work for you (it does for me).

cyclic99, 2021-02-22:
Well, I was a little bit unsure of your PoE HAT comment at first as I had not realised there was a linkage.
However I assume there is commonality with regards to the fan op.
So I extrapolated your instruction, the download came and it completed suggesting a boot,  which I duly did.
The reboot all runs ok but the "case fan" start up issue is still present.
(currently no overclock i.e. @1500 temp trigger 70deg - rises to ~80 with occasional fan twitch and or creep
eventually arm throttles @~83deg down 1000)

Is there a check I can make to positively confirm that I carried out the request correctly?

cyclic99, 2021-02-22:
The only fan movement I see is sometimes a counter-clockwise twitch e.g. the tip of a blade moves say 1, 2 or 3mm.
Usually it springs back to the pre-twitch position. I.e. no net movement.
Occasionally it does not spring back - I called that fan can creep.

I do not see any twitch or creep if I use a separate unconnected fan to bring the temp below the fan threshold. 
So when all is cool the fan is stationary.

The fan does not have its movement obstructed.

\# grep -v -E "^(#.*)?[\r\n]*$" /boot/config.txt
dtparam=audio=on
dtoverlay=vc4-fkms-v3d
max_framebuffers=2
arm_64bit=1
[all]
dtoverlay=gpio-fan,gpiopin=14,temp=70000
\#dmesg | grep PWM
\#dmesg | grep gpio-fan
[    8.101631] gpio-fan gpio-fan@0: GPIO fan initialized
\# vcgencmd measure_clock arm; vcgencmd measure_temp
frequency(48)=1000212864
temp=84.7'C
\# cat /var/log/syslog | grep PWM
\# cat /var/log/syslog | grep gpio-fan
Feb 22 13:36:25 zzz kernel: [    7.410041] gpio-fan gpio-fan@0: GPIO fan initialized
Feb 22 13:59:33 zzz kernel: [    6.892210] gpio-fan gpio-fan@0: GPIO fan initialized
Feb 22 14:02:50 zzz kernel: [    8.101631] gpio-fan gpio-fan@0: GPIO fan initialized
\# \#Multiple entries match reboots

I have ordered some extra fans in case this fan is faulty.

pelwell, 2021-02-22:
Apologies - I'd never really noticed that the case fan is just driven from a regular GPIO (I don't have one) with no kind of speed control.

Running on a 64-bit kernel I can see the gpio-fan overlay and the underlying driver doing the right thing. With two windows open I run this in one:
```sh
while true; do vcgencmd measure_temp; raspi-gpio get 14; sleep 1; done
```
and this in the other:
```sh
while true; do true; done
```
The temperature slowly starts to rise:
```
temp=45.2'C
GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE
temp=45.2'C
GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE
temp=44.8'C
```
As the temperature cross 50C (I set a lower threshold in config.txt) you see the GPIO go high:
```
temp=49.6'C
GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE
temp=49.6'C
GPIO 14: level=1 fsel=1 func=OUTPUT pull=NONE
temp=50.1'C
GPIO 14: level=1 fsel=1 func=OUTPUT pull=NONE
```
I suggest you run `raspi-gpio get 14`:
1. If the pin is an output then the overlay is at least partially working. 
2. If the temperature is above the threshold and the level is 1 then the fan hardware or wiring is faulty.
3. If the temperature is above the threshold and the level is 0 then there is a software fault of some kind - see what the following command reports: `grep . /sys/class/thermal/*/*`

cyclic99, 2021-02-23:
Thank you for your ongoing work.
Further testing to demonstrate non startup of case fan has been untaken.
Summary:
Throughout  # raspi-gpio get 14 shows:   GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE 
New Observation:
The pi4 froze for approximately 60 secs just after going through the temp trigger point of 70deg
After the (indented) 60 seconds the pi4 resumed as though nothing had "happened".

Connected? or Something else?

++++++++++++++++++++++++++++++++++++
Timed Log of test follow including your requests
++++++++++++++++++++++++++++++++++++
```
# 
# ###############################################################
# # Start - Use an external fan NOT pi4 connected to cool all
# ############################################################### 
# 
# date ; vcgencmd measure_clock arm;  vcgencmd measure_temp;  raspi-gpio get 14;  grep . /sys/class/thermal/*/*               
Tue 23 Feb 09:53:59 GMT 2021
frequency(48)=1500398464
temp=65.7'C
GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE
/sys/class/thermal/cooling_device0/cur_state:0
/sys/class/thermal/cooling_device0/max_state:1
grep: /sys/class/thermal/cooling_device0/power: Is a directory
grep: /sys/class/thermal/cooling_device0/subsystem: Is a directory
/sys/class/thermal/cooling_device0/type:gpio-fan
/sys/class/thermal/thermal_zone0/available_policies:step_wise 
grep: /sys/class/thermal/thermal_zone0/cdev0: Is a directory
/sys/class/thermal/thermal_zone0/cdev0_trip_point:0
/sys/class/thermal/thermal_zone0/cdev0_weight:0
grep: /sys/class/thermal/thermal_zone0/hwmon0: Is a directory
/sys/class/thermal/thermal_zone0/integral_cutoff:0
/sys/class/thermal/thermal_zone0/k_d:0
/sys/class/thermal/thermal_zone0/k_i:0
/sys/class/thermal/thermal_zone0/k_po:0
/sys/class/thermal/thermal_zone0/k_pu:0
/sys/class/thermal/thermal_zone0/mode:enabled
/sys/class/thermal/thermal_zone0/offset:410040
/sys/class/thermal/thermal_zone0/passive:0
/sys/class/thermal/thermal_zone0/policy:step_wise
grep: /sys/class/thermal/thermal_zone0/power: Is a directory
/sys/class/thermal/thermal_zone0/slope:-487
grep: /sys/class/thermal/thermal_zone0/subsystem: Is a directory
/sys/class/thermal/thermal_zone0/sustainable_power:0
/sys/class/thermal/thermal_zone0/temp:65731
/sys/class/thermal/thermal_zone0/trip_point_0_hyst:10000
/sys/class/thermal/thermal_zone0/trip_point_0_temp:70000
/sys/class/thermal/thermal_zone0/trip_point_0_type:active
/sys/class/thermal/thermal_zone0/type:cpu-thermal
#
# ############################################################### 
# Other Ext Fan off to allow app running to slowly heat (~4mins)
# ############################################################### 
# 
# date ; vcgencmd measure_clock arm;  vcgencmd measure_temp;  raspi-gpio get 14;  grep . /sys/class/thermal/*/*               
Tue 23 Feb 09:57:19 GMT 2021
frequency(48)=1500398464
temp=71.5'C
GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE
/sys/class/thermal/cooling_device0/cur_state:1
/sys/class/thermal/cooling_device0/max_state:1
grep: /sys/class/thermal/cooling_device0/power: Is a directory
grep: /sys/class/thermal/cooling_device0/subsystem: Is a directory
/sys/class/thermal/cooling_device0/type:gpio-fan
/sys/class/thermal/thermal_zone0/available_policies:step_wise 
grep: /sys/class/thermal/thermal_zone0/cdev0: Is a directory
/sys/class/thermal/thermal_zone0/cdev0_trip_point:0
[... 121 lines cut ...]
grep: /sys/class/thermal/thermal_zone0/cdev0: Is a directory
/sys/class/thermal/thermal_zone0/cdev0_trip_point:0
/sys/class/thermal/thermal_zone0/cdev0_weight:0
grep: /sys/class/thermal/thermal_zone0/hwmon0: Is a directory
/sys/class/thermal/thermal_zone0/integral_cutoff:0
/sys/class/thermal/thermal_zone0/k_d:0
/sys/class/thermal/thermal_zone0/k_i:0
/sys/class/thermal/thermal_zone0/k_po:0
/sys/class/thermal/thermal_zone0/k_pu:0
/sys/class/thermal/thermal_zone0/mode:enabled
/sys/class/thermal/thermal_zone0/offset:410040
/sys/class/thermal/thermal_zone0/passive:0
/sys/class/thermal/thermal_zone0/policy:step_wise
grep: /sys/class/thermal/thermal_zone0/power: Is a directory
/sys/class/thermal/thermal_zone0/slope:-487
grep: /sys/class/thermal/thermal_zone0/subsystem: Is a directory
/sys/class/thermal/thermal_zone0/sustainable_power:0
/sys/class/thermal/thermal_zone0/temp:64757
/sys/class/thermal/thermal_zone0/trip_point_0_hyst:10000
/sys/class/thermal/thermal_zone0/trip_point_0_temp:70000
/sys/class/thermal/thermal_zone0/trip_point_0_type:active
/sys/class/thermal/thermal_zone0/type:cpu-thermal
# ###############################################################
# 
```

pelwell, 2021-02-23:
Please repeat your test with the fan disconnected. A feature of `raspi-gpio` is that it reports the actual level observed at the pin, even when it is configured as an output. This means that even though the pin may be configured as an output driving high, the level may be displayed as 0 if there is a short to ground. (The reason I'm suspicious is that the `cooling_device0` section shows cur_state switch to 1 as you would expect but the GPIO not changing)

cyclic99, 2021-02-23:
I have completely removed the fan.
Run loop:
`while true ; do vcgencmd measure_clock arm;  vcgencmd measure_temp;  raspi-gpio get 14;  sleep 1; done`
Receive continuous:
`GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE`
(Visually watching level=0 from ~<60deg through 70deg to ~81deg  in and out of throttle)

Out of interest is there anything in the fan apart from the fan?    (You pick up +5 _and_ 0v)
I.e. is there a transistor in there or do you just rely on the pin drive?

lurch, 2021-02-23:
> Out of interest is there anything in the fan apart from the fan? (You pick up +5 and 0v)
I.e. is there a transistor in there or do you just rely on the pin drive?

https://www.raspberrypi.org/products/raspberry-pi-4-case-fan/ says:
* Input voltage: 5V DC supplied via 40-pin GPIO header on Raspberry Pi 4
* Fan speed control: Pulse width modulation control via user-selectable GPIO pin

pelwell, 2021-02-23:
I'm beginning to run out of ideas, but try changing the gpiopin value 14 to something else - 15 say, or 21, and watch that GPIO instead. It won't spin the fan, of course, but you ought to see the level changing.

pelwell, 2021-02-23:
Wait - before you do that, just try "raspi-gpio set 14 op dh; raspi-gpio get 14" and see if the fan spins.

cyclic99, 2021-02-24:
```
# raspi-gpio set 14 op dh; raspi-gpio get 14
GPIO 14: level=1 fsel=1 func=OUTPUT pull=NONE
```
fan spins for about 1 or 2 seconds - consistently. 
Also
Fan also spins with power only i.e. blue NOT connected to gpio14 
Also 
```
# raspi-gpio set 14 op dh; raspi-gpio get 14; raspi-gpio set 14 op dh; raspi-gpio get 14;   \
raspi-gpio get 14; sleep 2; raspi-gpio get 14
GPIO 14: level=1 fsel=1 func=OUTPUT pull=NONE
GPIO 14: level=1 fsel=1 func=OUTPUT pull=NONE
GPIO 14: level=1 fsel=1 func=OUTPUT pull=NONE
GPIO 14: level=0 fsel=1 func=OUTPUT pull=NONE
```
Correlates
Progress?

pelwell, 2021-02-24:
I haven't managed to get hold of a case fan yet (staff working from home), but substituting an LED for the fan on GPIO 14 shows it to be working as I would expect - it's off until the temperature crosses the threshold, then remains on until the temperature drops 10 degrees below that threshold. There's no kind of PWM speed control in operation, but a different fan driver might be able to do PWM. Switching between 32-bit and 64-bit makes no difference.

lurch, 2021-02-24:
IIRC GPIO14 is also one of the UART pins. @pelwell Is it possible that something's trying to write to the UART, and 'overriding' what the fan-driver software is trying to set? (that also might explain the "glitchy" behaviour being seen?)

pelwell, 2021-02-24:
As long as the GPIO is configured as an output (which it is) the UART can have no influence on it.

cyclic99, 2021-02-24:
I have more fans being delivered today for testing though after the last tests I do not now think it likely that the fan is faulty given that there is correlation been "level=..." and the fan "on/off".

My thoughts now after all the tests was that there might be 2 pieces of code driving the GPIO14.
So that when the trigger turns it on "some other dude" is turning it off again soon afterwards?

Now its all open, I could hang a scope on it?
I would expect that to show a short "on pulse" or with "raspi-gpio set 14 op dh" a longer pulse.

Bye the way,
Why does "... dh" it make the fan consistently work even if only for 2 or 3 seconds.

pelwell, 2021-02-24:
That's a timely comment - the supplies cupboard is unlocked and I have a case fan to test.

From what I can see the blue wire acts as a 3.3V gate to the 5V supply - high means the fan is on, and low means it is off. There must be a built-in pull-up that causes disconnected to mean on, which is what you saw. A shell script with `set 14 op dh` and `set 14 op dl` in a loop gives a slow spin - it's simply about duty cycle and the characteristics  of the motor.

Here's another experiment. Comment out the dtoverlay line in config.txt and reboot, then see if you can simply switch it on and off with raspi-gpio. If you find that the fan switches off after a while then that would point the finger towards some kind of contention for control of the GPIO.

cyclic99, 2021-02-24:
Before I changed anything,
I put a scope GPIO14, trigger set 70, temp above 70.
No fan connected
I see normally 0 volts but a bit high speed noisy (likely my poor earth connection) zero
Scope triggers on burst of nice clean of 3v3 pulses every 3 to 5 secs
Pulse period in burst ~10uS, looks like groupings of 8, and, a la Jonny 5, "but resembles SPI CLK"
Probably 12 bytes -  burst total ~1mS, occasionally a double burst

With "raspi-gpio set 14 op dh" one gets both a long positive pulse from ....dh as expected as well
but I also see the  pulse train busts as before.
The first burst after the ....dh terminates that long pulse i.e leaving GOPIO14 low
The appearance is not that of a electrical short, rather driven from single drive  (GPIO14)

Test with temp below the temp trigger point 70deg 
So with the temp down at 55deg with another fan
The bursts continue 

With "**#**dtoverlay=gpio-fan,gpiopin=14,temp=70000"
The bursts continue.
So one might conclude that gpio-fan is off the hook ?

In fact it seems there is a conflict of usage of GPIO 14
and some other "unknown" (to me) dude is pulsing GPIO14 from within ?

pelwell, 2021-02-24:
You could confirm that the fan and driver work by moving the blue wire to a different GPIO - the next ping along is 18 - and changing the dtoverlay line to match.

cyclic99, 2021-02-24:
The next pin is 10 (GPIO15) and looked active - high with occasional low going pulse - so I left it alone !!
The next after is pin 12 (GPIO18) permanently high and is the one I think you meant - so I tried it.
Using application generated self heat alternating and with external fan for cooling I checked pin with scope **no fan**.
The pin output on the scope correlate with the CLI level indication well.

I also checked with the same fan - also similar correlation.
I expect your LED would also correlate

For a more realistic use test I set up config.txt and rebooted:
```
dtoverlay=gpio-fan,gpiopin=18,temp=80000
arm_freq=2000
over_voltage=6
```
The **logic** of gpio-fan worked. 
i.e. as temp goes up the fan comes off, as temp goes down fan goes off
However using:   "vcgencmd measure_temp" as my metric
going up - should switch on at 80C but in fact switches on at ~65C 
going down - should switch off at 70C but it switches off at ~55C

So the 10degC hysteresis seems correct but the absolute **temperature** points  are ~15C low...
Curious?

PS
My box of test case fans has just arrived with postie . . .   :-)

pelwell, 2021-02-24:
Did you try any temperatures other than 80C?

My prediction is that all the fans behave the same as the one you've been trying - they work (modulo your incorrect threshold) on GPIOs other than 14.

cyclic99, 2021-02-24:
Agreed, hence :-)
Are there any other temps you would like to suggest that would be helpful to you?

pelwell, 2021-02-24:
70C seemed to work before. But use the entries in /sys/class/thermal to isolate where the threshold is failing. By the way, I've got 80C as a limit and the fan still hasn't kicked in at 73.5 (it's struggling to get higher at the moment - it needs a different workload).

cyclic99, 2021-02-24:
75, 78, 80, 90 all give 65/55 within measurable limits

"dtoverlay=gpio-fan,gpiopin=18,temp=80000"
Are the trailing 000 s relevant.   1/1000 of a degree ??
This temp parameter is either doing nothing or my format is wrong.

I have based my inputs on raspi-config which puts 80000 when requesting 80degC
70000, 80000, 90000

75.5degC :  Are you using 400 with new chip?  You seem to have a "high class problem"

cyclic99, 2021-02-24:
I have to go now; my conclusion is that using pios LITE
pin=18  with temp=80000 or whatever (70000 to 90000) has no effect on fan thresholds
at least with overclock 2000 and over voltage 6
pin=14 has others driving the pin preventing fan from working at all

pelwell, 2021-02-24:
> 75.5degC : Are you using 400 with new chip? You seem to have a "high class problem"

No, but it's a 4B with the BCM2711C0 fitted, and I didn't try very hard to warm it up.

pelwell, 2021-02-24:
For when you next look at this:
* `sudo vcdbg log msg` will show any errors from the firmware caused by trying to apply the overlay.
* `cat /sys/class/thermal/thermal_zone0/trip_*` will show if your chosen temperature threshold is being used.

wywywywy, 2021-02-25:
> Firmware release 99d9a483 was made over a week before the PoE HAT fix went in - running `sudo SKIP_KERNEL=1` should make it work for you (it does for me).

So is the new firmware going to be available for RPiOS64 as well?

By the way what's the `sudo SKIP_KERNEL=1` used in combination with?

cyclic99, 2021-02-25:
I have figured it out overnight.   It is not a rpi issue.
It was caused by residual 3rd party software left over from a previous install for a 3rd party fan being tested.
That is now uninstalled and rpi case fan works on gpio14 as it should.
Initial test indicates regulation seems to be at ~correct levels e.g. 78Cset -> On/Off 78/68.

closed 2021-02-25 by pelwell