raspberrypi/Raspberry-Pi-OS-64bit #137: Only low screen resolution (1024x768) on VNC

j64bit, 2021-02-23:
Hi,
just installed the 64bit image on two Raspi 3's. Works great (many thanks), but i found a little issue.

I use them headless via the raspi-config: Interface->P3 VNC->Yes setting.
Different to 32bit Pi OS, the VNC screen is only 1024x768 size. I'd like 1920x1028.

When HDMI is connected and at 1920x1024, then also VNC is. But in headless mode it is not available.

I read about "xandr" and scripts for increasing the screen. But maybe this should work out of the box and similar to 32bit Pi OS?

Any hints how to resolve?

Many thanks
Jörg

```
pi@weatherpi:~ $ xrandr 
Screen 0: minimum 320 x 200, current 1024 x 768, maximum 2048 x 2048
HDMI-1 connected primary 1024x768+0+0 (normal left inverted right x axis y axis) 0mm x 0mm
   1024x768      60.00* 
   800x600       60.32    56.25  
   848x480       60.00  
   640x480       59.94 
```

pelwell, 2021-02-23:
See the troubleshooting section on VNC's Raspberry Pi support page: https://help.realvnc.com/hc/en-us/articles/360002249917-VNC-Connect-and-Raspberry-Pi

j64bit, 2021-02-23:
Thanks! Works now.

Added this to /boot/config.txt from the troubleshooting support page.

```
hdmi_force_hotplug=1
hdmi_ignore_edid=0xa5000080
hdmi_group=2
hdmi_mode=82

```

closed 2021-02-23 by j64bit