raspberrypi/Raspberry-Pi-OS-64bit #139: Latest Raspberry OS - cannot change wallpaper

beng7, 2021-02-26:
Hi
As indicated in the title, I just cannot change the wallpaper in Preferences > Appearances settings > Desktop; any modification is just not registered.
And when I run `sudo pcmanfm` -w /usr/share/rpd-wallpaper/new_wallpaper.jpg`, it still does not do anything but report that the "Desktop manager is not active"...
Yet upon reboot I can see the wallpaper of the login screen is changed to the new_wallpaper.jpg but once logged in the session wallpaper remains the sad temple.jpg

I was able to cahnge the session wallpaper by renaming new_wallpaper.jpg to temple.jpg but on reboot I noticed the login screen wallpaper is now a light grey colour...

lurch, 2021-02-26:
This is the issue-tracker for Raspberry-Pi-OS-64bit. For troubleshooting issues with 32 bit Raspberry OS you should ask for help on the [forums](https://www.raspberrypi.org/forums/).

XECDesign, 2021-09-01:
Although nothing to do with 64-bit, you're seeing the issue because you're running the command as root.

closed 2021-09-01 by XECDesign