raspberrypi/Raspberry-Pi-OS-64bit #141: Several issues: screen blanking / audio (HDMI)  Pi4B rev 1.4

rkloost, 2021-03-09:
I've had several issues: audio not working, screen blanking not working.

Situation:
- flashed latest beta Raspberry OS to SD-card
- Booted raspberry pi 4b Rev1.4 8GB
- Standard config via GUI
- SSH & VNC enabled via GUI
- full update: sudo apt update ; sudo apt full-upgrade ; sudo apt autoremove
- reboot

Two issues:
Audio and screen blanking not working

After a lot of reading and trouble shooting:
- reinstall, the initial steps mentioned above...
- just making sure, configure the sound and screenblanking via 'sudo raspi-config' in a terminal.

And in a mysterious way now the issues are solved.

I suspect an issue (rights) with the GUI config tool. Anyway after the modifications on the CLI the 64-bit OS is running like a charm (for now that is).

Regards,
Ruud

BitBistro-code, 2021-04-06:
It is hard to tell from this description if this is really a bug. Have you tried getting support from the forums?

https://www.raspberrypi.org/forums/viewforum.php?f=28&sid=a50a74ffb1a74ba290881b490ce875dd

lurch, 2021-04-08:
@rkloost Please try reproducing this, and if you're still having problems please provide a detailed step-by-step reproduction walkthrough.

rkloost, 2021-04-12:
LS,

I was using the GUI to configure the initial install. After reinstalling I
used raspi-config for the configuration and it worked flawless.

lurch, 2021-04-12:
But IIRC the GUI tool simply calls the underlying CLI tool, which is why I was asking for a detailed (step by step) reproduction test case.