raspberrypi/Raspberry-Pi-OS-64bit #143: CEC devices (i.e. TV remotes for Kodi) do not work. Requesting a prebuild "Pulse-Eight / libcec" library be added to the repo

ioogithub, 2021-03-23:
CEC devices such as TV remotes do not currently work in Raspberry Pi OS 64bit so users cannot use TV remotes with Kodi.

The repo at (https://github.com/Pulse-Eight/libcec) contains code that works with 64-bit kms kernel. However the library needs to be built with this flag:

cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0

I am requesting that the library be prebuilt and included in the Raspberry Pi OS 64bit thus giving users the ability to use their CEC devices with Raspberry Pi OS 64bit.

The issue has been hashed out here:
https://github.com/Pulse-Eight/libcec/issues/543

milemar, 2021-06-15:
I had a similar problem with dietpi x64.
I have installed `cec-client -l` and got a reply:
`libvcos.so: cannot open shared object file: No such file or directory`

Apparently some symlinks are missing that are version independent so I ended up creating bunch of them:

`
ln -s libvcos.so.0 /usr/lib/aarch64-linux-gnu/libvcos.so
ln -s libvchiq_arm.so.0 /usr/lib/aarch64-linux-gnu/libvchiq_arm.so
ln -s libbcm_host.so.0 /usr/lib/aarch64-linux-gnu/libbcm_host.so
`
After that Kodi loded up CEC plugin successfully.

popcornmix, 2021-06-15:
@XECDesign if Raspberry-Pi-OS-64bit is kms by default then it makes sense to build with:
``cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=0``

XECDesign, 2021-06-15:
@popcornmix Does the same work for FKMS? Is it possible to support both?

popcornmix, 2021-06-15:
When I tested with:
``cmake -DHAVE_LINUX_API=1 -DHAVE_RPI_API=1``
the same library did work with kms and fkms for me (it attempts to connect both ways and only one should succeed).
However when I suggested this to a user he only got it working when built for the specific target.
IIRC this was on Arch and there could have been other reasons for the failure.

If you wanted to test:
```
cec-ctl -d0 --playback # Configure the PC as a CEC Playback device
cec-ctl -d0 -S # Show the CEC topology
```
should show if it's working. You'll need a CEC capable display (i.e. a TV rather than a monitor).

popcornmix, 2021-06-15:
Sorry that only tests linux cec. Running cec-client should indicate if it's working.

XECDesign, 2021-06-15:
I've updated libcec to 4.0.7 and enabled both options. If you edit `/etc/apt/sourced.list.d/raspi.list` to add `untested` after `main`, run `sudo apt update && sudo apt full-upgrade`, it should pull in the new version.

If that works as expected and doesn't break kodi, I can move it across to 'main' later.

popcornmix, 2021-06-16:
Okay. The updated version works on fkms, but not kms. The problem is:
```
pi@pi4:~ $ cec-client -l
libCEC version: 4.0.7, compiled on Linux-5.10.17-v8+ ... , features: P8_USB, DRM, P8_detect, randr, RPi, Exynos, Linux, AOCEC
Found devices: 2

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi

device:              2
com port:            Linux
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
```
and it tries using the RPi one by default.

I checked my own build and that was working correctly with both kms and fkms.

My build is using master of libcec plus this patch: https://github.com/Pulse-Eight/libcec/compare/master...Kwiboo:cec-framework-multiple.patch
That also has the benefit you can use `cec-client cec-client /dev/cec1` to use the second hdmi port's cec.

@Kwiboo are you able to PR your patch? It does appear to work with `/dev/cec1` as a destination, and also has the beneficial effect of not choosing the rpi interface when it's not available (but it still works when available with fkms).

popcornmix, 2021-06-16:
@Kwiboo there is a slight negative:
```
cec-client -l
libCEC version: 6.0.2, git revision: libcec-6.0.2+2-6239ae1, compiled on 2021-06-16 17:01:56 by pi@domnfs on Linux 5.10.43-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
ERROR:   [            3573]	error opening serial port '': No such file or directory
ERROR:   [            3573]	could not open a connection (try 1)
ERROR:   [            6409]	error opening serial port '': No such file or directory
ERROR:   [            6409]	could not open a connection (try 2)
ERROR:   [            8299]	error opening serial port '': No such file or directory
ERROR:   [            8299]	could not open a connection (try 3)
ERROR:   [            9559]	error opening serial port '': No such file or directory
ERROR:   [            9559]	could not open a connection (try 4)
Found devices: 3

device:              1
com port:            /dev/cec0
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

device:              2
com port:            /dev/cec1
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux

device:              3
com port:            
vendor id:           0000
product id:          0000
firmware version:    65535
```
when running with `-l` it slowly decides serial is not available. This isn't an issue when not using `-l`.
Without your patch serial isn't probed:
```
cec-client -l
libCEC version: 6.0.2, git revision: libcec-6.0.2+1-76551ea, compiled on 2021-06-16 17:18:10 by pi@domnfs on Linux 5.10.43-v7l+ (armv7l), features: P8_USB, DRM, P8_detect, randr, RPi, Linux
Found devices: 2

device:              1
com port:            RPI
vendor id:           2708
product id:          1001
firmware version:    1
type:                Raspberry Pi

device:              2
com port:            Linux
vendor id:           0000
product id:          0000
firmware version:    0
type:                Linux
```

XECDesign, 2021-06-18:
I've uploaded a build with Kwiboo's patch to `untested`.

XECDesign, 2021-06-22:
Let me know when it's safe to move to `main`.

popcornmix, 2021-06-24:
With latest update I could use cec-client on both kms and fkms (with default options), and cec worked by default with both kms and fkms on repo versions of kodi.

So I think this can move to main repo.

XECDesign, 2021-06-24:
Done

Marco-Total, 2021-11-17:
> I've updated libcec to 4.0.7 and enabled both options. If you edit `/etc/apt/sourced.list.d/raspi.list` to add `untested` after `main`, run `sudo apt update && sudo apt full-upgrade`, it should pull [...]

Hi,

I try your solution an add the untested channel to my raspi-apt-config:

```
/etc/apt/sourced.list.d/raspi.list
deb http://archive.raspberrypi.org/debian/ buster main untested
```

after run apt update && apt full-upgrade nothing displayed to update. I have the Problem, that I run kodi 19.1 under raspberry pi os 64bit, and cec not working:

```
root@mapi:/home/pi# uname -a
Linux mapi 5.10.63-v8+ #1459 SMP PREEMPT Wed Oct 6 16:42:49 BST 2021 aarch64 GNU/Linux
```

```
root@mapi24:/home/pi# cec-client -l
libCEC version: 4.0.5,  compiled on 2020-05-04 13:29:44 by pbuilder@guillaume.marillat.net on Linux 5.6.0-1-arm64 (aarch64), features: P8_USB, DRM, P8_detect, randr, Exynos, AOCEC
Found devices: NONE
root@mapi24:/home/pi#
```

```
root@mapi24:/home/pi# cec-client -m
starting a monitor-only client. use 'mon 0' to switch to normal mode
No device type given. Using 'recording device'
CEC Parser created - libCEC version 4.0.5
no serial port given. trying autodetect: FAILED
```

XECDesign, 2021-11-17:
The `untested` channel isn't needed anymore and I'd recommend removing it.

Your version of libcec doesn't seem to be what's currently in buster. What's the output of `apt policy libcec4`?

Marco-Total, 2021-11-18:
> 
> 
> The `untested` channel isn't needed anymore and I'd recommend removing it.
> 
> Your version of libcec doesn't seem to be what's currently in buster. What's the output of `apt policy libcec4`?

root@mapi:/home/pi# apt policy libcec4
libcec4:
  Installiert:           2:4.0.5-dmo0+deb10u1
  Installationskandidat: 2:4.0.5-dmo0+deb10u1
  Versionstabelle:
 *** 2:4.0.5-dmo0+deb10u1 100
        100 /var/lib/dpkg/status
     4.0.7+dfsg1-1+rpt2 500
        500 http://archive.raspberrypi.org/debian buster/main arm64 Packages
     4.0.4+dfsg1-2 500
        500 http://deb.debian.org/debian buster/main arm64 Packages

XECDesign, 2021-11-18:
Either you, or a script you ran, installed `version 2:4.0.5-dmo0+deb10u1` from an external source. The simplest thing to do would be to uninstall all libcec packages and reinstall again. If you have any trouble with that, it's best addressed on the forum.

Marco-Total, 2022-01-19:
Hi,

after update from buster to bulleseye the problem is done! The problem was, that i installed kodi 19.1 with libcec from untestet channel. In the bullseye repository is  a update to kodi 19.3 avalible, so cec working again.

Thx for your tipps!