raspberrypi/Raspberry-Pi-OS-64bit #144: Extreme lag in 4K

Dimension02000, 2021-03-25:
I have a RPI4 with 4GB RAM and whenever I have the resolution set to any 4K resolution I experience extreme lag. For example even simply moving the mouse is very slow. I tried to stream a 1080 video and it stuttered enough that I had to stop watching it. The stream was initially done using FireFox but since Chromium is the preferred browser for RPI I tried it but experienced the same issue. I have added in the `hdmi_enable_4kp60=1` to my /boot/config.txt file but this did not resolve the lag issue. I also confirmed that  my Samsung 55" TV is plugged into HDMI port 0.

Kuratius, 2021-04-09:
Are you using the 64 bit version? In that case swap to 32 bit, hardware acceleration is still broken for 64 bit afaik.

Kuratius, 2021-04-09:
In the 64 bit version hardware acceleration is still broken afaik, so this isn't anything new.

Dimension02000, 2021-04-09:
Is this even being considered with any up coming updates? Maybe this year?

Kuratius, 2021-04-09:
I think there's a (now unsupported because the maintainer signed on with a three letter agency or similar that doesn't allow them to work on other software projects, even in their spare time) gentoo fork that hacked hardware acceleration into their 64 bit version.
As far as I know the raspberry pi foundation is still trying to get the ones who make the dependencies for the hardware acceleration to agree on what standards/apis to use for their 64 bit version.

blacklat, 2021-04-17:
I bought my rpi4 in 2019 (2 years ago) and there is still no official 64-bit stable OS, this is extremely shameful for the raspberry folks and for customers it is extremely frustrating and annoying. I have no other words to describe the raspberry coders than "incompetent". The folks at retro game systems are years ahead of the raspberry folks. This only shows that the raspberry team is composed only of people without technical knowledge, amateurs, cheap labor.
One thing has to be said and understood by them, this shit is not free, I paid for it and paid dearly in the pre-launch, we are consumers who pay their wages, the operating system is their obligation to be made available for free to everyone who purchased an rpi board. If your employees lack the intelligence to build a stable system, then send them away and hire qualified people. I bought a 64-bit rpi board, fuck the 32-bit OS, make the fucking 64-bit OS work, be it professional, lamers.

edmundlaugasson, 2021-09-26:
I have actually same issue with RP4, 8 GB RAM. I [compiled Vulkan driver](https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=289168), then at least Full HD videos work. Not tested 4K yet. I guess that Vulkan driver should work initially as built-in. Currently needed to recompile on software update if related packages are updated.

chris-y, 2021-12-10:
I see this issue too. It's fine at every other resolution - including the one just below - so it does seem 4K specific.