raspberrypi/Raspberry-Pi-OS-64bit #145: USB

sgb999, 2021-04-11:
So i am not using the 64 bit version anymore because of this issue, when i was using it, it was last month and the issue is the USB drives keep disconnecting and reconnecting every few minutes or so, i had to change back to the stable 32 bit version because i was wanting to use openmediavault.

DiomedesDominguez, 2021-04-14:
@sgb999 can your please share which version/batch are you using? Maybe is a version compatibility issue, for example, if you are using the early batch of 1GB of RAM.

sgb999, 2021-04-14:
It was a 2gb RAM model that i got brand new from pi hut in january last year

lurch, 2021-04-14:
@sgb999 Without much more detailed info (e.g. size and types of USB drives connected, kernel logs, etc.) I doubt there's much that the Raspberry Pi engineers can do to investigate this.

sgb999, 2021-04-14:
Well i doubt im the only person out there with this issue

lurch, 2021-04-14:
I don't recall seeing any other complaints about it in this issue-tracker...

DiomedesDominguez, 2021-04-14:
> @sgb999 Without much more detailed info (e.g. size and types of USB drives connected, kernel logs, etc.) I doubt there's much that the Raspberry Pi engineers can do to investigate this.

@sgb999 , if possible, please share your kernel logs.

Fahrek, 2021-06-14:
I have the same issue. When I plug in the sandisk ultra flair 64gb pendrive in a raspi 4 with 8gb usb... well, openmediavault freeze the os

lurch, 2021-06-14:
> the USB drives keep disconnecting and reconnecting every few minutes or so

In hindsight, I wonder if that might be an insufficient-power problem? Might work more reliably if you connect your USB drives to a powered USB hub?

shenwii, 2021-09-15:
I have the same issue.

Model: raspi 4 with 2gb
USB3.0 port: an extra powered removable HDD drive
USB2.0 port: Wireless keyboard and mouse receiver

My case: the removable HDD drive works normally, but the wireless keyboard and mouse keep disconnecting and reconnecting every few minutes.

I found 2 logs, not sure if they help.

```
Sep 14 23:04:30 raspberrypi kernel: usb 1-1.3: USB disconnect, device number 3
Sep 14 23:04:30 raspberrypi thd[476]: Error reading device '/dev/input/event1'
Sep 14 23:04:30 raspberrypi thd[476]: Error reading device '/dev/input/event0'
Sep 14 23:04:45 raspberrypi kernel: usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
Sep 14 23:04:45 raspberrypi kernel: usb 1-1.3: New USB device found, idVendor=045e, idProduct=028e, bcdDevice= 1.05
Sep 14 23:04:45 raspberrypi kernel: usb 1-1.3: New USB device strings: Mfr=0, Product=2, SerialNumber=3
Sep 14 23:04:45 raspberrypi kernel: usb 1-1.3: Product: Controller
Sep 14 23:04:45 raspberrypi kernel: usb 1-1.3: SerialNumber: 00000000
Sep 14 23:04:45 raspberrypi mtp-probe[1520]: checking bus 1, device 4: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Sep 14 23:04:45 raspberrypi mtp-probe[1520]: bus: 1, device: 4 was not an MTP device
Sep 14 23:04:45 raspberrypi kernel: input: Microsoft X-Box 360 pad as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/input/input24
Sep 14 23:04:45 raspberrypi kernel: usbcore: registered new interface driver xpad
Sep 14 23:04:45 raspberrypi mtp-probe[1525]: checking bus 1, device 4: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Sep 14 23:04:45 raspberrypi mtp-probe[1525]: bus: 1, device: 4 was not an MTP device
```

```
Sep 14 23:05:27 raspberrypi kernel: usb 1-1.3: USB disconnect, device number 4
Sep 14 23:05:27 raspberrypi kernel: xpad 1-1.3:1.0: xpad_try_sending_next_out_packet - usb_submit_urb failed with result -19
Sep 14 23:05:42 raspberrypi kernel: usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
Sep 14 23:05:42 raspberrypi kernel: usb 1-1.3: New USB device found, idVendor=046d, idProduct=c52b, bcdDevice=12.03
Sep 14 23:05:42 raspberrypi kernel: usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Sep 14 23:05:42 raspberrypi kernel: usb 1-1.3: Product: USB Receiver
Sep 14 23:05:42 raspberrypi kernel: usb 1-1.3: Manufacturer: Logitech
Sep 14 23:05:42 raspberrypi kernel: logitech-djreceiver 0003:046D:C52B.0008: hiddev96,hidraw0: USB HID v1.11 Device [Logitech USB Receiver] on usb-0000:01:00.0-1.3/input2
Sep 14 23:05:42 raspberrypi mtp-probe[1559]: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Sep 14 23:05:42 raspberrypi mtp-probe[1559]: bus: 1, device: 5 was not an MTP device
Sep 14 23:05:42 raspberrypi kernel: input: Logitech M310/M310t as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.2/0003:046D:C52B.0008/0003:046D:4031.0009/input/input25
Sep 14 23:05:42 raspberrypi kernel: logitech-hidpp-device 0003:046D:4031.0009: input,hidraw1: USB HID v1.11 Mouse [Logitech M310/M310t] on usb-0000:01:00.0-1.3/input2:1
Sep 14 23:05:42 raspberrypi kernel: input: Logitech K520 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.2/0003:046D:C52B.0008/0003:046D:2011.000A/input/input26
Sep 14 23:05:42 raspberrypi kernel: logitech-hidpp-device 0003:046D:2011.000A: input,hidraw2: USB HID v1.11 Keyboard [Logitech K520] on usb-0000:01:00.0-1.3/input2:2
Sep 14 23:05:42 raspberrypi mtp-probe[1562]: checking bus 1, device 5: "/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3"
Sep 14 23:05:42 raspberrypi mtp-probe[1562]: bus: 1, device: 5 was not an MTP device
Sep 14 23:05:42 raspberrypi systemd-udevd[1557]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 14 23:05:42 raspberrypi systemd-udevd[1560]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 14 23:05:42 raspberrypi systemd-udevd[1568]: Process '/usr/sbin/th-cmd --socket /var/run/thd.socket --passfd --udev' failed with exit code 1.
Sep 14 23:05:42 raspberrypi systemd-logind[475]: Watching system buttons on /dev/input/event1 (Logitech K520)
Sep 14 23:05:44 raspberrypi kernel: logitech-hidpp-device 0003:046D:4031.0009: HID++ 2.0 device connected.
Sep 14 23:05:45 raspberrypi kernel: logitech-hidpp-device 0003:046D:2011.000A: HID++ 1.0 device connected.
```