raspberrypi/Raspberry-Pi-OS-64bit #147: Can't access the raspberry first time burn

pifparfait, 2021-04-13:
After burning the device, I have internet and I use the "Cyberduck" to connect as a VNC client to the Raspberry. But from the terminal, I can access the raspberry with raspberry as the password. Any help?

sudo ssh -t  pi@192.168.1.52 -p 22 "cd \/boot && exec \$SHELL -l"
Password:
Sorry, try again.
Password:
Sorry, try again.
Password:
sudo: 3 incorrect password attempts

pelwell, 2021-04-13:
https://www.raspberrypi.org/documentation/remote-access/ssh/

lurch, 2021-04-13:
There's no need to use `sudo` for ssh - I suspect in your output above you're actually being asked for your sudo password rather than the ssh password :wink:

pifparfait, 2021-04-13:
Hi @lurch, I'm asking for the ssh.  I burnt before one 32 bits with the same Raspberry. I changed the SD card with the 64 bits, and I got this message: @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ECDSA key sent by the remote host is
SHA256:7Q2AiidytG6Hrg9xdJOPesYfxpPtj/SMHQ2yFKRweIA.
Please contact your system administrator.
Add correct host key in /Users/parfait/.ssh/known_hosts to get rid of this message.
Offending ED25519 key in /Users/parfait/.ssh/known_hosts:4
ECDSA host key for 192.168.1.52 has changed and you have requested strict checking.
Host key verification failed. 

So I'm trying to use the sudo to control ssh

pifparfait, 2021-04-13:
@pelwell I'm using the Raspberry without a screen. I guess I need to access first by command line to the device, then execute sudo raspi-config. 
My problem is that I can't actually access it...

lurch, 2021-04-13:
> REMOTE HOST IDENTIFICATION HAS CHANGED

That's because your local ssh client has "remembered" the ECDSA key from your 32-bit installation of Raspberry Pi OS, and this is different to the ECDSA key on your 64-bit installation of Raspberry Pi OS (this file is generated the very first time Raspberry Pi OS boots up). As the error-message suggests, to fix it you need to delete line 4 of `/Users/parfait/.ssh/known_hosts`. And then when you connect to SSH on your 64-bit installation of Raspberry Pi OS (without using sudo), your local ssh client will remember this "new" ECDSA host key instead.

pifparfait, 2021-04-13:
Amazing!! many thanks!!!

pifparfait, 2021-04-13:
Another question before closing this. @lurch, I heard that the 64-bit doesn't have VNC server. It is correct?

lurch, 2021-04-13:
> I heard that the 64-bit doesn't have VNC server. It is correct?

https://github.com/raspberrypi/Raspberry-Pi-OS-64bit/issues?q=is%3Aissue+vnc+is%3Aclosed suggests that the problems with VNC have been fixed? (although you might need to do a `sudo apt update && sudo apt upgrade` first? :shrug: )

pifparfait, 2021-04-13:
Let me check it! many thanks @lurch !!!