raspberrypi/Raspberry-Pi-OS-64bit #148: pyscf cython for 64bit

pifparfait, 2021-04-13:
Hi, Do you know if we have a version of pyscf cython for 64bits?

pifparfait, 2021-04-13:
ERROR: Command errored out with exit status 1:
     command: /home/pi/rasqberry/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qtbjvxc9/pyscf_868400215376487690dfdb4ce41d2d17/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qtbjvxc9/pyscf_868400215376487690dfdb4ce41d2d17/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-nx2p6wna
         cwd: /tmp/pip-install-qtbjvxc9/pyscf_868400215376487690dfdb4ce41d2d17/
    Complete output (12 lines):
    ****************************************************************
    *** WARNING: BLAS library not found.
    * You can include the BLAS library in the global environment LDFLAGS, eg
    *   export LDFLAGS='-L/path/to/blas/lib -lblas'
    * or specify the BLAS library path in  PYSCF_INC_DIR
    *   export PYSCF_INC_DIR=/path/to/blas/lib:/path/to/other/lib
    ****************************************************************
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-qtbjvxc9/pyscf_868400215376487690dfdb4ce41d2d17/setup.py", line 245, in <module>
        raise RuntimeError
    RuntimeError
    ----------------------------------------

pifparfait, 2021-04-13:
WARNING: Discarding https://files.pythonhosted.org/packages/cb/00/2c660e993fe7b333bd776ab281298ba033297ddd715cf99479043b288ba1/pyscf-1.7.6.tar.gz#sha256=4a51c070e032d53cdb2a4a9ab0bb2f296abb3d14b92c9b1ad35d695e95f5c9b3 (from https://pypi.org/simple/pyscf/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Downloading pyscf-1.7.5.tar.gz (7.6 MB)
     |████████████████████████████████| 7.6 MB 263 kB/s 
    ERROR: Command errored out with exit status 1:
     command: /home/pi/rasqberry/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qtbjvxc9/pyscf_82151e75093046d385283cb843474701/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qtbjvxc9/pyscf_82151e75093046d385283cb843474701/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-j90w5rqc
         cwd: /tmp/pip-install-qtbjvxc9/pyscf_82151e75093046d385283cb843474701/
    Complete output (12 lines):
    ****************************************************************
    *** WARNING: BLAS library not found.
    * You can include the BLAS library in the global environment LDFLAGS, eg
    *   export LDFLAGS='-L/path/to/blas/lib -lblas'
    * or specify the BLAS library path in  PYSCF_INC_DIR
    *   export PYSCF_INC_DIR=/path/to/blas/lib:/path/to/other/lib
    ****************************************************************
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-qtbjvxc9/pyscf_82151e75093046d385283cb843474701/setup.py", line 245, in <module>
        raise RuntimeError
    RuntimeError
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/62/be/87e1f4aea51181b25f71cec4c9a60f60cbebf26ca17d17a5b153dd22a8d9/pyscf-1.7.5.tar.gz#sha256=52856b39f0ada2f6340757caa65dc5c1d9a3cdfceea2a6615ad8af92664a6c69 (from https://pypi.org/simple/pyscf/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Downloading pyscf-1.7.4.tar.gz (7.5 MB)
     |████████████████████████████████| 7.5 MB 9.5 MB/s 
    ERROR: Command errored out with exit status 1:
     command: /home/pi/rasqberry/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qtbjvxc9/pyscf_b34b326075254feaae889f33ea6ef10f/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qtbjvxc9/pyscf_b34b326075254feaae889f33ea6ef10f/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-1owa8owk
         cwd: /tmp/pip-install-qtbjvxc9/pyscf_b34b326075254feaae889f33ea6ef10f/
    Complete output (12 lines):
    ****************************************************************
    *** WARNING: BLAS library not found.
    * You can include the BLAS library in the global environment LDFLAGS, eg
    *   export LDFLAGS='-L/path/to/blas/lib -lblas'
    * or specify the BLAS library path in  PYSCF_INC_DIR
    *   export PYSCF_INC_DIR=/path/to/blas/lib:/path/to/other/lib
    ****************************************************************
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-qtbjvxc9/pyscf_b34b326075254feaae889f33ea6ef10f/setup.py", line 245, in <module>
        raise RuntimeError
    RuntimeError
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/c0/f1/fe52a94e92acf3fa8c5543d86cfed6fea0526011d2ff2dda352f9cf9eaef/pyscf-1.7.4.tar.gz#sha256=a7734863f660dca275aabb633f8afc86b6bdc6ba6332a4841202fdc0f67e79e8 (from https://pypi.org/simple/pyscf/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Downloading pyscf-1.7.3.tar.gz (23.3 MB)
     |████████████████████████████████| 23.3 MB 25 kB/s 
    ERROR: Command errored out with exit status 1:
     command: /home/pi/rasqberry/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-qtbjvxc9/pyscf_cbf8e7eea8a14f149e3d04daf4f1b349/setup.py'"'"'; __file__='"'"'/tmp/pip-install-qtbjvxc9/pyscf_cbf8e7eea8a14f149e3d04daf4f1b349/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-14x180e4
         cwd: /tmp/pip-install-qtbjvxc9/pyscf_cbf8e7eea8a14f149e3d04daf4f1b349/
    Complete output (12 lines):
    ****************************************************************
    *** WARNING: BLAS library not found.
    * You can include the BLAS library in the global environment LDFLAGS, eg
    *   export LDFLAGS='-L/path/to/blas/lib -lblas'
    * or specify the BLAS library path in  PYSCF_INC_DIR
    *   export PYSCF_INC_DIR=/path/to/blas/lib:/path/to/other/lib
    ****************************************************************
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-qtbjvxc9/pyscf_cbf8e7eea8a14f149e3d04daf4f1b349/setup.py", line 245, in <module>
        raise RuntimeError
    RuntimeError
    ----------------------------------------
[... cut ...]
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/a7/5d/6991b173ab24f35c453f5adca3d6829566730abb3b3b571913517303be24/pyscf-1.4.0.zip#sha256=471f53181e511c869b0d162ffb754cca58f282b86e33f0dc3883461e26553ad3 (from https://pypi.org/simple/pyscf/). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement pyscf
ERROR: No matching distribution found for pyscf

lurch, 2021-04-13:
This sounds like a support-question for the pyscf maintainers rather the Raspberry Pi OS maintainers?

pifparfait, 2021-04-13:
@lurch many thanks! do you have the link to the Raspberry Pi OS maintaniners?

lurch, 2021-04-13:
I think you misunderstood my comment. I'm suggesting you should get in touch with the pyscf maintainers. (I know nothing about pyscf, I've never heard of it before you mentioned it)

pifparfait, 2021-04-13:
Got it! 
many thanks @lurch

closed 2021-04-13 by pifparfait