raspberrypi/Raspberry-Pi-OS-64bit #149: DNS issue, resolv.conf doesn't reflect interfaces configurations

craph, 2021-04-16:
Hi,

I open this issue because I have a weird behavior on my Raspberry Pi 4 8Gb where I have installed [raspios_arm64-2020-08-24](https://downloads.raspberrypi.org/raspios_arm64/images/raspios_arm64-2020-08-24/)

After I did the installation, I made changes in `/etc/network/interfaces.d/eth0`  like this : 
```bash
cat /etc/network/interfaces.d/eth0
auto eth0
allow-hotplug eth0
iface eth0 inet static
        address 192.168.0.200
        netmask 255.255.255.0
        gateway 192.168.0.1
        dns-nameservers 127.0.0.1 1.1.1.1
```

I put **dns-nameservers** to `127.0.0.1` (because I have my own setup for dns-over-https and I want to have a second one I case I have an issue that's why I put `1.1.1.1`.

But in the **resolv.conf** file I can't see that. I have only the first value like that : 
```bash
cat /etc/resolv.conf
# Generated by resolvconf
nameserver 127.0.0.1
```

I don't understand why I don't have a second line with the second dns entry like this : 
```bash
cat /etc/resolv.conf
# Generated by resolvconf
nameserver 127.0.0.1
nameserver 1.1.1.1
```

What am I doing wrong ?

lurch, 2021-04-16:
Are you sure that this is specific to Raspberry Pi OS 64bit and doesn't also affect Raspberry Pi OS 32bit?
I think this might be a question for the https://www.raspberrypi.org/forums/ rather than GitHub? :man_shrugging:

craph, 2021-04-16:
Hi @lurch ,

Thank you very much for your reply. I have also open a discussion on the forum too but I think GitHub is also a good place to open an issue. 

I have also tested to replace **dns-nameservers** by **dns-nameserver** and add this twice (one for each entry) but it was the same result.

I suppose it's à real issue because the configuration I made is "compliant" with the man of interfaces.

lurch, 2021-04-16:
...but unless you also test on Raspberry Pi OS 32bit, we can't tell if it's a problem that _only_ affects Raspberry Pi OS 64bit (which is specifically what this repo is for, it's not for "general issues"), or if it's a problem with the upstream Debian package?
(if it's a general issue, it should instead be reported at https://github.com/RPi-Distro/repo/issues )

craph, 2021-04-18:
@lurch Thank you for the update.

I already tested it on Raspberry Pi OS 32bits too and I have the same issue. So I agree to open an issue for the 32bits version.

Can you give me the link where I should open the issue please ?

lurch, 2021-04-18:
"(if it's a general issue, it should instead be reported at https://github.com/RPi-Distro/repo/issues )" :smiley:

craph, 2021-04-20:
@lurch 
I have just tried to use dhcpcd.conf only and it's not working too.

```bash
pi@raspberrypi:~ $ cat /etc/dhcpcd.conf
# A sample configuration for dhcpcd.
# See dhcpcd.conf(5) for details.

# Allow users of this group to interact with dhcpcd via the control socket.
#controlgroup wheel

# Inform the DHCP server of our hostname for DDNS.
hostname

# Use the hardware address of the interface for the Client ID.
clientid
# or
# Use the same DUID + IAID as set in DHCPv6 for DHCPv4 ClientID as per RFC4361.
# Some non-RFC compliant DHCP servers do not reply with this set.
# In this case, comment out duid and enable clientid above.
#duid

# Persist interface configuration when dhcpcd exits.
persistent

# Rapid commit support.
# Safe to enable by default because it requires the equivalent option set
# on the server to actually work.
option rapid_commit

# A list of options to request from the DHCP server.
option domain_name_servers, domain_name, domain_search, host_name
option classless_static_routes
# Respect the network MTU. This is applied to DHCP routes.
option interface_mtu

# Most distributions have NTP support.
#option ntp_servers

# A ServerID is required by RFC2131.
require dhcp_server_identifier

# Generate SLAAC address using the Hardware Address of the interface
#slaac hwaddr
# OR generate Stable Private IPv6 Addresses based from the DUID
slaac private

# Example static IP configuration:
#interface eth0
#static ip_address=192.168.0.10/24
#static ip6_address=fd51:42f8:caae:d92e::ff/64
#static routers=192.168.0.1
#static domain_name_servers=192.168.0.1 8.8.8.8 fd51:42f8:caae:d92e::1
interface eth0
static ip_address=192.168.0.245/24
static routers=192.168.0.1
static domain_name_servers=127.0.0.1 1.1.1.1

# It is possible to fall back to a static IP if DHCP fails:
# define static profile
#profile static_eth0
#static ip_address=192.168.1.23/24
#static routers=192.168.1.1
#static domain_name_servers=192.168.1.1

# fallback to static profile on eth0
#interface eth0
#fallback static_eth0
```

And the resolv.conf : 
```bash
pi@raspberrypi:~ $ cat /etc/resolv.conf
# Generated by resolvconf
nameserver 127.0.0.1
```

Why I can't see the second DNS server mentionned in dhcpcd.conf ?