raspberrypi/Raspberry-Pi-OS-64bit #152: Wifi driver unstable, brcmf_sdio_txfail: sdio error

SMH17, 2021-05-07:
Sometimes a crash of Wifi occurr. 

This problem happens randomly. (I encountered it during the upload from an external drive.)

Here the log.
```
[ 3053.918262] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
[ 3053.918281] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[ 3053.918812] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000038
[ 3053.918832] Mem abort info:
[ 3053.918847]   ESR = 0x96000005
[ 3053.918864]   EC = 0x25: DABT (current EL), IL = 32 bits
[ 3053.918878]   SET = 0, FnV = 0
[ 3053.918892]   EA = 0, S1PTW = 0
[ 3053.918906] Data abort info:
[ 3053.918920]   ISV = 0, ISS = 0x00000005
[ 3053.918934]   CM = 0, WnR = 0
[ 3053.918959] user pgtable: 4k pages, 39-bit VAs, pgdp=000000010b836000
[ 3053.918974] [0000000000000038] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
[ 3053.919021] Internal error: Oops: 96000005 [#1] PREEMPT SMP
[ 3053.919033] Modules linked in: rfcomm bnep hci_uart btbcm bluetooth ecdh_generic ecc squashfs 8021q garp stp llc binfmt_misc fuse sg vc4 brcmfmac cec brcmutil v3d drm_kms_helper gpu_sched bcm2835_isp(C) snd_soc_core bcm2835_v4l2(C) drm bcm2835_codec(C) cfg80211 bcm2835_mmal_vchiq(C) drm_panel_orientation_quirks snd_compress videobuf2_vmalloc rfkill snd_pcm_dmaengine snd_bcm2835(C) v4l2_mem2mem videobuf2_dma_contig snd_pcm videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon snd_timer videobuf2_common snd videodev syscopyarea sysfillrect sysimgblt vc_sm_cma(C) mc fb_sys_fops backlight rpivid_mem uio_pdrv_genirq uio nvmem_rmem ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter i2c_dev nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp aes_neon_bs nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables sha256_generic nfnetlink aes_neon_blk crypto_simd cryptd
[ 3053.919462]  ip_tables x_tables ipv6
[ 3053.919500] CPU: 1 PID: 75 Comm: kworker/u8:2 Tainted: G         C        5.10.17-v8+ #1403
[ 3053.919510] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[ 3053.919568] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
[ 3053.919586] pstate: 60000005 (nZCv daif -PAN -UAO -TCO BTYPE=--)
[ 3053.919618] pc : bcm2835_mmc_transfer_dma+0x98/0x230
[ 3053.919632] lr : bcm2835_mmc_transfer_dma+0x88/0x230
[ 3053.919641] sp : ffffffc011adb8e0
[ 3053.919650] x29: ffffffc011adb8e0 x28: 000000000000000f 
[ 3053.919669] x27: 0000000000005a00 x26: 0000000000000600 
[ 3053.919687] x25: ffffffc011adbd50 x24: 0000000000000600 
[ 3053.919705] x23: 000000000000000f x22: ffffffc011adbb20 
[ 3053.919722] x21: 0000000000000001 x20: ffffff8100959880 
[ 3053.919739] x19: ffffff8101532540 x18: 0000000000000001 
[ 3053.919756] x17: 0000000000000000 x16: 0000000000000000 
[ 3053.919773] x15: ffffffffffffffff x14: 4c6dd6cecb26f127 
[ 3053.919796] x13: 632d2e91b59c7915 x12: f1d8af61cbd72c84 
[ 3053.919813] x11: 512118ffe4bd3a1a x10: 598c215e765fc5f5 
[ 3053.919830] x9 : ffffffc0101117a8 x8 : 3178197777710508 
[ 3053.919847] x7 : 8cdc1db881d910f6 x6 : ffffff8039ccc600 
[ 3053.919864] x5 : 0000000000000000 x4 : 0000000000000000 
[ 3053.919880] x3 : 000000000000003f x2 : 509c72acde731000 
[ 3053.919897] x1 : 0000000000000000 x0 : ffffff810091bc80 
[ 3053.919915] Call trace:
[ 3053.919930]  bcm2835_mmc_transfer_dma+0x98/0x230
[ 3053.919943]  bcm2835_mmc_request+0xa4/0xc0
[ 3053.919959]  __mmc_start_request+0x84/0x1e8
[ 3053.919972]  mmc_start_request+0x8c/0xb8
[ 3053.919985]  mmc_wait_for_req+0x74/0xf8
[ 3053.920025]  mmc_submit_one.isra.12+0x90/0x120 [brcmfmac]
[ 3053.920065]  brcmf_sdiod_sglist_rw+0x230/0x510 [brcmfmac]
[ 3053.920108]  brcmf_sdiod_send_pkt+0x110/0x130 [brcmfmac]
[ 3053.920146]  brcmf_sdio_dataworker+0x1810/0x22b8 [brcmfmac]
[ 3053.920160]  process_one_work+0x1f4/0x4d8
[ 3053.920170]  worker_thread+0x50/0x480
[ 3053.920184]  kthread+0x148/0x158
[ 3053.920196]  ret_from_fork+0x10/0x30
[ 3053.920212] Code: 2a0003f7 34000980 f9407a61 f9400280 (f9401c21) 
[ 3053.920226] ---[ end trace a56352f3039202e9 ]---
```

pelwell, 2021-05-07:
Does your external drive have its own power supply? The WiFi interface seems to be one of the things that suffers first when the voltage drops.

SMH17, 2021-05-07:
@pelwell 
Only related to video overscan, out resolution and HDMI, but Rapberry was in headless mode without any display attached when the issue happened. 
The software was updated to latest available 12h ago.

pelwell, 2022-05-30:
A hardware checksum error suggests bad signal integrity, which could be caused by:
* a very electrically noisy environment (this seems unlikely),
* an inadequate power supply (or poor power cable), or
* a hardware fault.
Can you try with a powered USB hub?

c10l, 2022-06-06:
Thanks for your comment. I tried the only powered hub I had at hand but it didn't provide the Pi with enough voltage, so I went back to the other PSU and so far it's been fine! :shrug:

lurch, 2022-06-06:
I suspect pelwell meant connect the powered hub between the Pi and the 3D-printer, rather than powering the Pi itself from the USB hub?

pelwell, 2022-06-06:
Indeed - the idea was to take some load off the Pi's PSU, but it sounds like the problem has been resolved.

c10l, 2022-06-06:
Ah got it. I used to run it over a different hub but that's now occupied in another place. I tried to isolate the 5V pin on the USB cable but then the Pi wouldn't recognise the serial port. Probably a quirk of the Einsy board Prusa uses as I had a similar setup in a different printer.

I also don't want to run a hub between the Pi and the printer as it would add considerable bulk to the setup (I would as a troubleshooting step though, of course).

So far it's been fine though, so 🤞 . :)

matthewlloyd, 2022-08-03:
I have been having this issue on and off for the last year on a 4B and have tried almost everything to fix it. I have some observations. For me it does seem to be power related:

* I have had the same issue with the official Raspberry Pi 5.1V power supply connected via USB-C, and a Meanwell 75W 5V supply running at 5.1V over AWG 22 wire soldered directly to the probe points on the Pi.
* Adding over_voltage=1 to /boot/firmware/config.txt did seem to reduce the frequency of this error to the point where I went a few weeks without it, compared to multiple times a week at first.
* Normally my Pi is located very close to the router. I can reliably reproduce this issue by moving it into the next room then connecting to my webcam stream, suggesting that one trigger for the issue is the BCM chip bumping into a higher transmit power mode due to poor reception.

It's possible over_voltage=2 or higher will reduce occurrences still further but I haven't tried that yet.

pelwell, 2022-08-05:
FYI we consider `over_voltage=2` a safe minimal over-voltage. Feel free to bump your current setting.

matthewlloyd, 2022-08-25:
> FYI we consider `over_voltage=2` a safe minimal over-voltage. Feel free to bump your current setting.

Thank you, I have been running at over_voltage=2 for a couple of weeks now, and saw the same issue again today. The pattern was the same as before - I had no crashes until I moved the Pi into the next room where the WiFi signal is weak, after which it lasted only a couple of hours. However this time I wasn't even accessing the connection, so the bandwidth use would have been very low. The crash caused a kernel panic and system shutdown.

It's as if the Broadcom chip can freeze when using high transmit power levels. I guess it could be overheating - I have a full-board heatsink with fan but checking the instructions it seems they provided thermal pads for everything except the WiFi module - or it could be a power distribution issue. For now I have disabled the WiFi entirely, but I may try reducing the max transmit power from the default 31dBm (1 Watt) with something like `iwconfig wlan0 txpower 15`.

```
Aug 25 05:29:34 pi kernel: [219833.319724] mmc1: Timeout waiting for hardware interrupt.
Aug 25 05:29:34 pi kernel: [219833.325460] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Aug 25 05:29:34 pi kernel: [219833.333033] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Aug 25 05:29:44 pi kernel: [219843.559636] mmc1: Timeout waiting for hardware interrupt.
Aug 25 05:29:44 pi kernel: [219843.565858] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Aug 25 05:29:44 pi kernel: [219843.573316] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Aug 25 05:29:44 pi kernel: [219843.582925] Unable to handle kernel read from unreadable memory at virtual address 0000000000000038
Aug 25 05:29:44 pi kernel: [219843.593503] Mem abort info:
Aug 25 05:29:44 pi kernel: [219843.596988]   ESR = 0x96000004
Aug 25 05:29:44 pi kernel: [219843.601654]   EC = 0x25: DABT (current EL), IL = 32 bits
Aug 25 05:29:44 pi kernel: [219843.607498]   SET = 0, FnV = 0
Aug 25 05:29:44 pi kernel: [219843.610738]   EA = 0, S1PTW = 0
Aug 25 05:29:45 pi kernel: [219843.614050]   FSC = 0x04: level 0 translation fault
Aug 25 05:29:45 pi kernel: [219843.619227] Data abort info:
Aug 25 05:29:45 pi kernel: [219843.622377]   ISV = 0, ISS = 0x00000004
Aug 25 05:29:45 pi kernel: [219843.626460]   CM = 0, WnR = 0
Aug 25 05:29:45 pi kernel: [219843.629591] user pgtable: 4k pages, 48-bit VAs, pgdp=00000000403e1000
Aug 25 05:29:45 pi kernel: [219843.636234] [0000000000000038] pgd=0000000000000000, p4d=0000000000000000
Aug 25 05:29:45 pi kernel: [219843.643236] Internal error: Oops: 96000004 [#1] PREEMPT SMP
Aug 25 05:29:45 pi kernel: [219843.648978] Modules linked in: cmac algif_hash algif_skcipher af_alg bnep hci_uart btqca btrtl btbcm btintel tls wireguard libchacha20poly1305 chacha_neon poly1305_neon libblake2s libcurve25519_generic libchacha libblake2s_generic ip6_udp_tunnel udp_tunnel nft_counter nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_comment nft_compat nf_tables binfmt_misc nfnetlink btsdio bluetooth ecdh_generic ecc brcmfmac brcmutil 88x2bu(OE) raspberrypi_hwmon bcm2835_v4l2(C) uvcvideo snd_usb_audio bcm2835_isp(C) bcm2835_codec(C) cfg80211 snd_hwdep snd_usbmidi_lib bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc snd_rawmidi snd_seq_device cdc_acm videobuf2_memops snd_bcm2835(C) snd_pcm videobuf2_v4l2 videobuf2_common snd_timer videodev snd vc_sm_cma(C) mc bcm2835_gpiomem rpivid_mem nvmem_rmem uio_pdrv_genirq gpio_fan uio sch_fq_codel dm_multipath scsi_dh_rdac scsi_dh_emc scsi_dh_alua ipmi_devintf ipmi_msghandler mtd pstore_blk usb_f_ecm drm
Aug 25 05:29:45 pi kernel: [219843.649125]  ramoops g_ether usb_f_rndis pstore_zone reed_solomon u_ether libcomposite efi_pstore ip_tables x_tables autofs4 btrfs blake2b_generic zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor xor_neon raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid uas usb_storage spidev crct10dif_ce dwc2 roles i2c_bcm2835 udc_core spi_bcm2835 xhci_pci xhci_pci_renesas phy_generic aes_arm64
Aug 25 05:29:45 pi kernel: [219843.777037] CPU: 2 PID: 1097327 Comm: kworker/u9:3 Tainted: G         C OE     5.15.0-1012-raspi #14-Ubuntu
Aug 25 05:29:45 pi kernel: [219843.787010] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
Aug 25 05:29:45 pi kernel: [219843.793014] Workqueue: brcmf_wq/mmc1:0001:1 brcmf_sdio_dataworker [brcmfmac]
Aug 25 05:29:45 pi kernel: [219843.800300] pstate: 20400005 (nzCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Aug 25 05:29:45 pi kernel: [219843.807449] pc : bcm2835_mmc_transfer_dma+0x98/0x23c
Aug 25 05:29:45 pi kernel: [219843.812584] lr : bcm2835_mmc_transfer_dma+0x88/0x23c
Aug 25 05:29:45 pi kernel: [219843.817710] sp : ffff8000121cb910
Aug 25 05:29:45 pi kernel: [219843.821158] x29: ffff8000121cb910 x28: 00000000000006b8 x27: ffff8000121cbb50
Aug 25 05:29:45 pi kernel: [219843.828500] x26: ffff40498d75bc00 x25: ffff8000121cbb08 x24: 0000000000001a00
Aug 25 05:29:45 pi kernel: [219843.835842] x23: 0000000000000001 x22: 0000000000000006 x21: ffff404982d45800
Aug 25 05:29:45 pi kernel: [219843.843182] x20: ffff404980212a80 x19: ffff404982d45e00 x18: 0000000000000000
Aug 25 05:29:45 pi kernel: [219843.850519] x17: 696d72657420646e x16: ffffdf06f58f15f4 x15: ffff404987f02f00
[... cut ...]
Aug 25 05:29:45 pi kernel: [219843.945639]  brcmf_sdio_dataworker+0x4c/0x90 [brcmfmac]
Aug 25 05:29:45 pi kernel: [219843.951046]  process_one_work+0x204/0x4c0
Aug 25 05:29:45 pi kernel: [219843.955200]  worker_thread+0x144/0x470
Aug 25 05:29:45 pi kernel: [219843.959085]  kthread+0x12c/0x140
Aug 25 05:29:45 pi kernel: [219843.962441]  ret_from_fork+0x10/0x20
Aug 25 05:29:45 pi kernel: [219843.966154] Code: 2a0003f6 340009c0 f9407a61 f9400280 (f9401c21)
Aug 25 05:29:45 pi kernel: [219843.972421] ---[ end trace b052618564b5b767 ]---
```

lurch, 2022-08-26:
I wonder if the "full-board heatsink" might be partially blocking the signal from the Wifi antenna? (although I'm not sure if that'd be the cause of your kernel crash :shrug: )

MayPLaY, 2022-11-01:
It looks like I'm having the same problem. In my case I'm using my Pi4 as a NAS. When retrieving large pics over WiFi through my editing app, my Pi just get unresponsive all of sudden. If I go check directly on my Pi through, the top-right WiFi icon is just out of service and is replaced "Network not found" icon.

Here is the log from /var/log/syslog:
```
Oct 31 21:30:30 raspberrypi kernel: [ 8318.658732] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:30:30 raspberrypi kernel: [ 8318.658886] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:30:30 raspberrypi kernel: [ 8318.658914] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:30:40 raspberrypi kernel: [ 8328.898881] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:30:40 raspberrypi kernel: [ 8328.898988] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:30:40 raspberrypi kernel: [ 8328.899012] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:30:51 raspberrypi kernel: [ 8339.139040] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:30:51 raspberrypi kernel: [ 8339.139137] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:30:51 raspberrypi kernel: [ 8339.139164] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:01 raspberrypi kernel: [ 8349.379186] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:31:01 raspberrypi kernel: [ 8349.379316] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:31:01 raspberrypi kernel: [ 8349.379345] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:11 raspberrypi kernel: [ 8359.619340] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:31:11 raspberrypi kernel: [ 8359.619489] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:31:11 raspberrypi kernel: [ 8359.619518] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:21 raspberrypi kernel: [ 8369.859487] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:31:21 raspberrypi kernel: [ 8369.859568] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:31:21 raspberrypi kernel: [ 8369.859592] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:32 raspberrypi kernel: [ 8380.099643] mmc1: Timeout waiting for hardware interrupt.
Oct 31 21:31:32 raspberrypi kernel: [ 8380.099796] brcmfmac: mmc_submit_one: CMD53 sg block write failed -110
Oct 31 21:31:32 raspberrypi kernel: [ 8380.099824] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:32 raspberrypi kernel: [ 8380.100335] brcmfmac: mmc_submit_one: CMD53 sg block write failed -84
Oct 31 21:31:32 raspberrypi kernel: [ 8380.100359] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:32 raspberrypi kernel: [ 8380.740022] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:35 raspberrypi kernel: [ 8383.470819] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:36 raspberrypi kernel: [ 8384.500102] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:36 raspberrypi kernel: [ 8384.826066] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:37 raspberrypi kernel: [ 8385.111299] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:37 raspberrypi kernel: [ 8385.540078] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:38 raspberrypi kernel: [ 8386.581338] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:39 raspberrypi kernel: [ 8387.620085] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:40 raspberrypi kernel: [ 8388.660096] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:42 raspberrypi kernel: [ 8390.580551] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:43 raspberrypi kernel: [ 8391.620135] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:44 raspberrypi kernel: [ 8392.660135] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:45 raspberrypi kernel: [ 8393.700958] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:46 raspberrypi kernel: [ 8394.740299] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:46 raspberrypi kernel: [ 8394.827121] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:47 raspberrypi kernel: [ 8395.780318] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:48 raspberrypi kernel: [ 8396.822156] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:49 raspberrypi kernel: [ 8397.860231] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:50 raspberrypi kernel: [ 8398.900205] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:31:52 raspberrypi kernel: [ 8399.941829] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
[... cut ...]
Oct 31 21:32:21 raspberrypi kernel: [ 8429.060686] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Oct 31 21:32:22 raspberrypi kernel: [ 8430.100702] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
```

Rebooting my Pi4 just fix the problem until the next time.

Here is my kit if it can be of any help: [https://www.amazon.ca/gp/product/B07YCWRZ8P/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1](https://www.amazon.ca/gp/product/B07YCWRZ8P/ref=ppx_yo_dt_b_asin_title_o07_s00?ie=UTF8&psc=1)

Thanks

p.s. The fan is non-stop running.

isegal, 2022-11-02:
Same problem here. Raspberry Pi 4 Model B Rev 1.1
Device became unstable after switching from ethernet to WiFi.
Appears to occur when transferring or seeking large files over the network.

Problem with the stock firmware:
```
Firmware: BCM4345/6 wl0: Jan  4 2021 19:56:29 version 7.45.229
```

Tried with the newer (cypress) firmware, problem persists:
```
Firmware: BCM4345/6 wl0: Apr 15 2021 03:03:20 version 7.45.234 (4ca95bb CY) FWID 01-996384e2
```

Setting `over_voltage=2` did not help. `txpower 15` also did not help.

Similar messages in the kernel log when the issue occues:
```
Nov  2 00:05:43 pi kernel: [1474514.651516] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:06:44 pi kernel: [1474575.712839] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:06:45 pi kernel: [1474576.732435] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:06:46 pi kernel: [1474577.772456] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:06:47 pi kernel: [1474579.128849] brcmfmac: mmc_submit_one: CMD53 sg block write failed -84
Nov  2 00:06:47 pi kernel: [1474579.128879] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:07:47 pi kernel: [1474638.977911] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:07:48 pi kernel: [1474640.013391] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
Nov  2 00:07:49 pi kernel: [1474641.053400] brcmfmac: brcmf_sdio_txfail: sdio error, abort command and terminate frame
```

The interesting pattern here is an error from mmc as well, so it might be hardware-related.
Using aftermarket power supply and a powered USB hub for disk drives.