raspberrypi/Raspberry-Pi-OS-64bit #153: install openjdk-8-jre in Raspberry Pi OS 64bit

bugalo, 2021-05-07:
I would like to install ImageJ in Raspberry Pi OS 64bit:

https://imagej.net/Raspberry_Pi

But it requires openjdk-8-jre, which is not in the repository. Could you, please, add it to the repository?

Thanks.

rag2, 2021-05-08:
That page dates from 2019, pre Pi OS Buster. However, ImageJ itself 
appears to support OpenJDK 11, as suggested at 
https://imagej.nih.gov/ij/notes.html.

You might, probably should, be able to get ImageJ to run under the 
provided OpenJDK.

Good luck

PS: Open JDK 8 falls out of active support (bar security) in 10 months' 
time.

bugalo, 2021-05-13:
Thanks a lot for your responses.

I managed to download java 8 ARM 64-bit from Oracle's webpage, install it in Raspberry Pi OS 64bit, and have ImageJ running. Now that I have ImageJ running, I agree with you that I need to make it run under OpenJDK 11.