raspberrypi/Raspberry-Pi-OS-64bit #154: brcmfmac: brcmf_parse_vndr_ies: invalid vndr ie. length is too small 4

b3nn0, 2021-05-15:
When using the PI4 (with raspios aarch64 lite) as a Wifi P2P-GO (Wifi Direct),
the line `[   60.941291] brcmfmac: brcmf_parse_vndr_ies: invalid vndr ie. length is too small 4` is spammed in the kernel log about twice per second. Didn't observe that with previous 32 bit based versions:
[image]

pelwell, 2021-05-15:
Have you got any version numbers for the non-spammy 32- and spammy 64-bit kernels?

b3nn0, 2021-05-15:
Actually.. I guess I should have tested more before. Recent 32bit kernels do this as well apparently.
Not spammy 32 bit kernel:
```
Linux raspberrypi 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux
```

Spammy 32bit/64bit kernels:
Linux raspberrypi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
Linux raspberrypi 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux

So not really a 64bit issue, just a general kernel issue.