raspberrypi/Raspberry-Pi-OS-64bit #155: WiFi strange

Anenemy1, 2021-06-02:
Hello, I dont know if i'm writing in the right place. First of all i have a raspbbery pi 4b 8GB. I have installed pi os 64bit and i use it for home pc mostly. Today when i turned on i cant connect with wifi and after a few reboots and I delete and remaked of wpa_supplicant.conf file which nothing happen. I connected through ethernet to trubleshoot. Instead I checked for updates for the reason "if an update solve this i am nice." Nope. After that I connected manually through raspi-config. My question is: why in desktop in status line i see the symbol of ethernet(the two opposite vectors)? Did i make something wrong somewhere? I never had such problem before. i used pi os arm64 for 1 year i think.

lurch, 2021-06-02:
Possibly a question for @spl237 ?

Anenemy1, 2021-06-02:
ok i 'll write to him.

Anenemy1, 2021-06-02:
Sorry, i am new to github how i write to him?

spl237, 2021-06-02:
I'm spl237...

We haven't, to my knowledge, changed the 64-bit image for a year now, so if something isn't working, it's probably caused by a local change you made, or your card becoming corrupt. I'd suggest you flash a clean copy of the 27/5 image onto a new card and see if the problem goes away.

The 64-bit image is still a beta; there's no real need for anyone to be using it unless they are running one of a very small number of applications (mostly big databases) which need to address more than 4GB of RAM. For the vast majority of people, you'd be better off using the standard 32-bit image, which is regularly updated and fixed.

Anenemy1, 2021-06-02:
Ok. Thanks for the fast responce. I 'll do what you say. I 'll make backups from the folders of apps on pi directory and install a fresh copy of 32bit OS.

Anenemy1, 2021-06-02:
Hello again. I installed the 32bit pi os as i said before. After auto resising filesystem and set a few things up with the preinstall helper (like if i show borders on screen, connect to wifi and change pass) i open a terminal and my commands are: sudo apt update
sudo apt full-upgrade
#ArgonOneCaseDriver
curl https://download.argon40.com/argon1.sh | bash
sudo rm /etc/sudoers.d/010_pi-nopasswd
sudo apt install p7zip-full
# Enable non-free source repository by editing /etc/apt/sources.list.
cat /etc/apt/sources.list
sudo apt-get update
mkdir ~/unrar-nonfree && cd ~/unrar-nonfree
sudo apt-get build-dep unrar-nonfree
sudo apt-get source -b unrar-nonfree
sudo dpkg -i unrar*.deb
cd && rm -r ~/unrar-nonfree
sudo apt install rpi-imager
sudo apt install firefox-esr
sudo apt install thunderbird
sudo apt remove qpdfview
sudo apt install xpdf
sudo apt install libreoffice
sudo apt install realvnc-vnc-viewer
sudo apt install npm
sudo npm install nativefier -g
sudo apt install neofetch
sudo apt-get install guvcview
sudo apt autoremove
sudo apt clean
sudo raspi-config
After a few reboots i am again in the same situation. It shows not connected in the status bar. I rewrite the wifi password and still no connection.

Anenemy1, 2021-06-03:
I solved it by messing with the argon one settings. I change the fan to full speed all the time and restore it back. Tested both 32bit and 64bit with success. Sorry for wasting your time for a problem I myself solve.

lurch, 2021-06-03:
That's an awful lot of steps, and it's impossible to know which one might be causing your problem. The best thing to do would be to do each step one at a time, and then test after each step, to help narrow down which step is the cause of your problem.

closed 2021-06-12 by Anenemy1