raspberrypi/Raspberry-Pi-OS-64bit #158: A2DP - does it work on 64 bit?

MarkDH102, 2021-06-16:
I have a pair of Sony SBH80 bluetooth headphones (with MIC).
I have an up to date 64 bit Beta install on a Pi4 8GB.
I followed this advice :

**sudo apt update && sudo apt full-upgrade
sudo apt install pulseaudio-module-bluetooth pavucontrol
sudo apt autoremove
sudo reboot

pair your Bluetooth Audio Device, then right click on Audio Applet and choose BT Audio.

PAVUControl is a GUI for PulseAudio Sound Server.**

The headphones pair OK. BUT listening to music, I can only seem to use the HSP/HFP profile within the PAVUControl volume control applet. I can actually select the A2DP (Sink) option for high fidelity sound, but the sound quality stays exactly the same (i.e. low fidelity HSP/HFP).

lurch, 2021-06-16:
Does it work any differently on Raspberry Pi OS 32-bit?

ghollingworth, 2021-06-16:
Start with beta version and update.

Don't install anything else...

Pair using the Bluetooth logo in the panel bar (click Add Device)
Then right click the speaker symbol and select it under 'Audio Outputs' 

If you right click on the speaker symbol and select device profiles you'll see which ones it supports.

MarkDH102, 2021-06-16:
I did start off with the beta version. But I have a few packages installed. ZuluCrypt and Arduino IDE (1.8.12) with support for ESP32/ESP8266/DUE. Glade. CuteCOM. Okular (PDF viewer) and EasyTAG (mp3 tagger). Support for a P1005 HP printer.
It's basically my 'daily driver'. So going back to a fresh install will be a bit of an issue. 
I do however have a backup card from about 3 weeks ago that doesn't have the pulseaudio stuff installed. Worth trying that?
I did try connecting (and they appeared as a Bluetooth device) before installing pulseaudio etc. but couldn't see my device listed in the audio controls anywhere. Hence going to pulseaudio...

I don't have a 32 bit system that is connected to a screen (they are all via VNC). Would trying the headphones on one of those still work across VNC?

lurch, 2021-06-16:
> It's basically my 'daily driver'. So going back to a fresh install will be a bit of an issue.

Fortunately, new microSD cards are very cheap these days... :wink:

MarkDH102, 2021-06-18:
Ok. I put my old (updated today) 32 bit Buster install into my other Raspberry Pi 4 (1GB) and ran it headless (via VNC). It could not see ANY Bluetooth devices and my mobile phone couldn't see it.
Undeterred, I have now managed to burn a fresh 64 bit install (dated 28-5-21) and updated it. Same 1GB Pi4, headless using VNC. Again it can't see ANY devices on the Bluetooth and my mobile phone couldn't see it.
TBH I've not done much with the Bluetooth on Raspberry Pi's, but from this experience, I think that I'll be leaving it alone in the future... No more time to waste on this unfortunately.